USC-ACTLab/crazyswarm #244: How to properly set up motive streaming engine

swarm5, 2020-08-28:
I ran into problems with motive’s streaming engine settings:
1. Libobjecttracker: I have always been successful in this mode. My settings are as shown in the figure below, but when I roslanuch, **it will always be stuck at frame 0:**
[image]

> ========
> 
> PARAMETERS
>  * /mocap_helper/motion_capture_type: optitrack
>  * /mocap_helper/object_tracking_type: libobjecttracker
>  * /mocap_helper/optitrack_host_name: 192.168.1.10
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.15
> 
> NODES
>   /
>     mocap_helper (crazyswarm/mocap_helper)
> 
> auto-starting new master
> process[master]: started with pid [7454]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 8ea7922c-e930-11ea-9d67-b42e99612c7c
> process[rosout-1]: started with pid [7476]
> started core service [/rosout]
> process[mocap_helper-2]: started with pid [7486]
> frame 0:
> 

2. MotionCapture: Unlike many people, I only have to set labeled markers to on for my data to be normal. However, the flight is not stable now, I don't know if it is the problem of streaming engine settings.
I want to know if anyone can give me a reference?

whoenig, 2020-08-29:
1. Usually that means that your IP address is off or you have some other connectivity issues. There also seems to be a mismatch between "10.0.0.2" and "192.168.1.10".
2. A good way to test it is: 1) run mocap_helper, manually move the drone and make sure the output makes sense, 2) run hover_swarm with logging enabled (log stateEstimate.x, stateEstimate.y, stateEstimate.z), move the drone manually (do not fly), and monitor the state estimate. The first test makes sure your system is properly configured, the second test ensures your various firmwares operate correctly. For example, if you did not update your radio or NRF firmwares, the second test might fail.

If you have trouble with the connectivity, troubleshoot with the standalone test application from https://github.com/whoenig/NatNetSDKCrossplatform.

swarm5, 2020-08-29:
I use motive2.0, and use NatNetSDKCrossplatform to view the output data, the results are as follows:
1. Libobjecttracker: No output
[image]
`$ ./mcr 192.168.1.10`

> NatNetVersion: 80.145.47.0
> ServerVersion: 3.0.0.0
> 

2. MotionCapture: there is output
[image]

```
> 
> > Begin Packet
> > -------
> > Message ID : 7
> > Byte count : 116
> > Frame # : 2905618
> > Marker Set Count : 0
> > Unidentified Marker Count : 0
> > Rigid Body Count : 1
> > ID : 1
> > pos: [0.00,-0.02,0.02]
> > ori: [-0.00,0.00,0.00,-1.00]
> > Mean marker error: 0.00
> > Skeleton Count : 0
> > Labeled Marker Count : 0
> > Timestamp : 24213.483
> > Mid-exposure timestamp : 27005018398
> > Camera data received timestamp : 27005028728
> > Transmit timestamp : 27005031639
> > End Packet
> > -------------
> > Begin Packet
> > -------
> > Message ID : 7
> > Byte count : 116
> > Frame # : 2905619
> > Marker Set Count : 0
> > Unidentified Marker Count : 0
> > Rigid Body Count : 1
> > ID : 1
> > pos: [0.00,-0.02,0.02]
> > ori: [-0.00,0.00,0.00,-1.00]
> > Mean marker error: 0.00
> > Skeleton Count : 0
> > Labeled Marker Count : 0
> > Timestamp : 24213.492
> > Mid-exposure timestamp : 27005044376
> > Camera data received timestamp : 27005054833
> > Transmit timestamp : 27005057631
> > End Packet
> > -------------
> > Begin Packet
> > -------
> > Message ID : 7
> > Byte count : 116
> > Frame # : 2905620
> > Marker Set Count : 0
> > Unidentified Marker Count : 0
> > Rigid Body Count : 1
> > ID : 1
> > pos: [0.00,-0.02,0.02]
> > ori: [-0.00,0.00,0.00,-1.00]
> > Mean marker error: 0.00
> > Skeleton Count : 0
> > Labeled Marker Count : 0
> > Timestamp : 24213.500
> > Mid-exposure timestamp : 27005070351
> > Camera data received timestamp : 27005081351
> > Transmit timestamp : 27005086673
> > End Packet
> > -------------
> > Begin Packet
> > -------
> > Message ID : 7
> > Byte count : 116
> > Frame # : 2905621
> > Marker Set Count : 0
> > Unidentified Marker Count : 0
> > Rigid Body Count : 1
> > ID : 1
> > pos: [0.00,-0.02,0.02]
> > ori: [-0.00,0.00,0.00,-1.00]
> > Mean marker error: 0.00
> > Skeleton Count : 0
> > Labeled Marker Count : 0
> > Timestamp : 24213.508
> > Mid-exposure timestamp : 27005096329
> > Camera data received timestamp : 27005107139
> > Transmit timestamp : 27005112835
> > End Packet
> > -------------
> > Begin Packet
> > -------
> > Message ID : 7
> > Byte count : 116
> > Frame # : 2905622
> > Marker Set Count : 0
> > Unidentified Marker Count : 0
> > Rigid Body Count : 1
> > ID : 1
> > pos: [0.00,-0.02,0.02]
> > ori: [-0.00,0.00,0.00,-1.00]
> > Mean marker error: 0.00
> > Skeleton Count : 0
> > Labeled Marker Count : 0
> > Timestamp : 24213.517
> > Mid-exposure timestamp : 27005122308
> > Camera data received timestamp : 27005132832
> > Transmit timestamp : 27005138072
> > End Packet
> > 
```

swarm5, 2020-08-31:
After I upgraded my motive to 2.2.0 and maintained this setting, I solved my problem

closed 2020-08-31 by swarm5