USC-ACTLab/crazyswarm #245: crazyflie is unstable and often flips after taking off

swarm5, 2020-08-29:
I have encountered a serious problem, but I still don’t know the problem:
Recently, I cloned t`he new master-dfd69b4`. When I ran python niceHover.py, **I found out: When flying two cfs, one cf will always turn over; at the same time, when I test a single cf, it will also Discovered the phenomenon of flight instability and overturning.**
I suspect it is the latest master-dfd69b4 problem. I cloned an older `branch ros_time_abstraction` today. **This problem also exists. When I tested the single cf, I also found that the flight was unstable and flipped**.
1. There does not seem to be a problem with the data and positioning estimates of the motion capture system. I followed @whoenig 's tips and used NatNetSDKCrossplatform to get the data of the motion capture system. The data seemed to have no obvious problems. At the same time, I turned on logging and recorded the estimated value of xyz coordinates, which seems to be correct.
2. I suspect it is the problem of cf firmware, but I don't know why the same problem exists when I use different branches of crazyswarm, maybe this may also be a communication problem. **I upload a log of my running nicehover.py, can anyone help me analyze it.**
**Cf flips after hovering:**
[image]
```
roslaunch crazyswarm hover_swarm.launch
... logging to /home/ss/.ros/log/7d2ef0e2-ea0e-11ea-a663-200db01c05e8/roslaunch-407-19165.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35511/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: True
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
[... 149 lines cut ...]
[ WARN] [1598715855.226508278]: Latency too high! Is 0.017654 s.
[ INFO] [1598715861.133347997]: Takeoff!
[ INFO] [1598715863.662526160]: [cf1] GoTo
[ INFO] [1598715867.060084864]: NextPhase!
[ INFO] [1598715867.070247018]: Land!
[ WARN] [1598715868.886149911]: No updated pose for motion capture object cf1
[ INFO] [1598715869.084089777]: CF Console: SYS: ----------------------------
[ INFO] [1598715869.118583083]: CF Console: SYS: Crazyflie 2.1 is up and running!
[ INFO] [1598715869.141829917]: CF Console: SYS: Build 757:217cff5db4b2 (2016.09 +757) CLEAN
[ INFO] [1598715869.176499341]: CF Console: SYS: I am 0x203936564E4B501400200048 and I have 1024KB of flash!
[ INFO] [1598715869.188396121]: CF Console: CFGBLK: v1, verification [OK]
[ INFO] [1598715869.200080126]: CF Console: DECK_CORE: 1 deck(s) found
[ INFO] [1598715869.223598143]: CF Console: DECK_CORE: Calling INIT on driver bcLedRing for deck 0
[ INFO] [1598715870.126513628]: CF Console: IMU: BMI088 Gyro I2C connection [OK].
[ INFO] [1598715870.183363010]: CF Console: IMU: BMI088 Accel I2C connection [OK]
[ INFO] [1598715870.206584802]: CF Console: IMU: BMP388 I2C connection [OK]
[ INFO] [1598715870.230132433]: CF Console: ESTIMATOR: Using Complementary (1) estimator
[ INFO] [1598715870.253263578]: CF Console: CONTROLLER: Using PID (1) controller
[ INFO] [1598715870.722662752]: CF Console: EEPROM: I2C connection [OK].
[ INFO] [1598715871.537904803]: CF Console: SYS: The system resumed after watchdog timeout [WARNING]
[ INFO] [1598715871.561548518]: CF Console: SYS: Assert failed at src/drivers/src/nvic.c:142

```
Here is my flight log：
https://drive.google.com/drive/folders/14TvsCL2WbxcivZRMQHemH5bl7j9a83_r?usp=sharing

swarm5, 2020-08-30:
What I need to add is: For the latest version: new master-dfd69b4 I ran into the same problem as #182 .When cf rolls over, there will be the following error message: 

> Assert failed src/modules/src/crtp.c:186.

 I saw @whoenig 's answer, which seems to be a valid solution, but I still have a few questions.

1. After I comment `ASSERT(0);`, what else do I need to do? Do I need to re-build` ./build.sh` or re-upload the firmware?
2. How to upload cf firmware correctly? I use the ros command: `rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin `and `rosrun crazyflie_tools flash --target stm32 --filename prebuilt/cf2.bin`. Does this command upload the firmware of crazyswarm or the official firmware of cf? I often don’t know if the upload is successful, is there any tool for instructions?
3. My cf will restart after falling. Is it because I failed to upload the crazyswarm firmware?https://github.com/USC-ACTLab/crazyswarm/issues/240#issuecomment-668910103
4. The phenomenon of my 2019 branch `ros_time_abstractionversion` of crazyswarm seems to be similar to #64 . I want to know how it was solved in the end?
5. Is there a verified and stable branch? Can I achieve a stable flight of multiple cfs? I want to use speed vector control instructions or position control instructions. Do you have a recommended branch? @whoenig

swarm5, 2020-08-30:
In order to solve the problem of Assert failed src/modules/src/crtp.c:186, should I do this?
1. Comment out line 186 in `crazyflie-firmware/src/modules/src/crtp.c `under the crazyswarm project.
2. Use `./build.sh` to compile the firmware.
3. Use the ros command to upload the compiled firmware to cf.

whoenig, 2020-08-30:
1. In the `crazyflie-firmware` folder type `make`. Flash the firmware by clicking "Flash STM32" in chooser.py.
2. I recommend using the button in chooser.py, which internally will use `crazyflie_tools flash` with the firmware in the `crazyflie-firmware` folder (not the prebuilt one). Another option is to use `make flash` if you own a JTAG adapter. I do not recommend using `make cload` as this does not include a verification step. All other commands write a new firmware and verify the written bytes afterwards, reporting an error if the verification fails. You can use the latest official firmware, as long as you comment the `ASSERT(0);`.
3. Automatic restart of firmwares after failure is normal for newer firmware builds that have the watchdog enabled.
4./5. I recommend using the latest master branch. I recently used it with single-marker tracking with up to 16 CFs without issues.

swarm5, 2020-08-31:
@whoenig I modified and uploaded the firmware according to your instructions, but the phenomenon of rollover still occurs when I run it.
I recorded the flight log and found that the xy direction cannot be operated according to ctrltarget. I don't know what went wrong?
[image]
[image]
[image]

swarm5, 2020-08-31:
I upload all the logs of a flight, can anyone help me see where the problem is.
https://drive.google.com/drive/folders/1b-SeLKKiHuZswWjMErcplOfB4_Xljhlo?usp=sharing

swarm5, 2020-08-31:
My phenomenon is similar to the flying video in #150

whoenig, 2020-08-31:
Looks like a coordinate system mismatch as in #150. For the Crazyflie, positive x is the direction of the little antenna, positive z is upwards, and positive y is given by the right-hand-rule. Make sure that this matches what you get from Motive (no need to fly for this test). If the markers are small for your mocap space, there are also sometimes axis flips in Motive, which can cause erratic behavior. Switch to single-marker tracking in that case or put bigger markers in a unique configuration on the CF.

swarm5, 2020-08-31:
I used to log the marker points before, and the results seemed to be correct, and I checked the axis every time.
I flew four planes separately, two of them could fly but were unstable, and two of them would fall over repeatedly.
I use motive+optitrack, I don’t know how to set the tracking mode, I can only set the rigid body mode in motive.
In #244 , I asked about the tracking mode setting of motive.

whoenig, 2020-08-31:
Tracking Mode: In Motive disable/delete the object that you created for the CF. Then you should see 4+ markers which should be streamed as in #244. In hover_swarm, select "libobjecttracker" rather than "motionCapture" [here](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L36). As CF type use "defaultSingleMarker", see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L27. (Change the type in your allCrazyflies.yaml, then restart chooser.py, disable and re-enable all CFs). If you keep 4 markers, use "default" as type. However, if your Optitrack has trouble tracking, a single larger marker is often better.

I did check #244 and your settings and results look good (not sure what else you need in that thread?).

Did you check your latency? The Motive and Ubuntu PCs should be connected via Ethernet, not Wifi. You should avoid using a VM. You can check the latency between your Ubuntu machine and the CF using `rosrun crazyflie_tools comCheck --uri <your-URI>`, which should report a roundtrip latency of < 10ms.

swarm5, 2020-08-31:
1. I use a switch and Ethernet, but it seems that the delay is still very high. I don't know why.

> ss@407:~/crazyswarm$ rosrun crazyflie_tools comCheck --uri radio://0/100/2M/E7E7E7E70f
> numPacketsReceived: 74.5%
> Avg. roundtrip time: 26.3517 ms
> 736.377 packets/s sent
> link quality: 0.882667
> ss@407:~/crazyswarm$ rosrun crazyflie_tools comCheck --uri radio://0/100/2M/E7E7E7E70f
> numPacketsReceived: 94.5%
> Avg. roundtrip time: 22.0582 ms
> 708.215 packets/s sent
> link quality: 0.983333

2. The defaultSingleMarker has been used successfully. My solution is: upgrade motive to 2.2.0, and then only open labeled markers.

3. I regret that I still have problems. When I run nicehover.py, a phenomenon similar to #154 appeared. cf often ignores commands sent by the server, and it remains motionless in place. I tried to modify the channel. After modifying the channel, cf sometimes took off, but it still turned over.

4. I am looking for larger marking points for testing.

5. I would like to ask, what are the advantages and disadvantages of single-point tracking and four-point tracking? How does single-point tracking calculate attitude?

swarm5, 2020-08-31:
I would like to ask, Avg. roundtrip time, what is the roundtrip time measured? Is it the time between PA and cf communication?
What I need to explain is that my ubuntu and motive are connected through a switch using Ethernet.

swarm5, 2020-08-31:
What I need to add is:
When I change the antenna direction of the PA (perpendicular to the ground of the venue), the acceptance rate is greatly improved (~95%), but the round trip time is still very long (~20ms)

swarm5, 2020-08-31:
@whoenig  My current problem is still that the drone will flip after taking off. I think the problem should still be in the firmware:
1. My communication quality is 1
```
ss@407:~/crazyswarm$ rosrun crazyflie_tools comCheck --uri radio://0/100/2M/E7E7E7E70f
numPacketsReceived: 96%
Avg. roundtrip time: 21.2448 ms
709.723 packets/s sent
link quality: 1
```
2. I held cf with my hand and tested stateEstimate.x.y.x
The result of the log record is consistent with the path I walked.
[image]
3. Regarding the firmware, the version I am using is the latest version of matser. I have annotated the assertion according to your prompt, and made the firmware, and uploaded the firmware using chooser.py.
4. I have repeatedly confirmed the problem of the coordinate axis.

swarm5, 2020-08-31:
@whoenig 
This is my latest test result:
1. I changed the big mark, cf flying is particularly stable! ! ! ! ! Thank you very much! ! ! !
2. But I still have a problem: When I fly two planes, the indicator light of cf2 will flash red and green, and when I run nicehover.py, only cf1 can fly.
What I need to add is: sometimes, cf2 sometimes has this error message, this is a firmware problem, but I have uploaded the firmware using chooser.py: 

> [INFO] [1598875360.778264192]: [cf2] OW: Cmd 0x21 timeout.<F>

swarm5, 2020-09-01:
I want to share two phenomena:
1. When the terminal displays [cf2] OW: Cmd 0x21 timeout error, sometimes I can make the plane take off.
2. Existing problem: When I try to fly two cfs, the LED of cf2 will always flash red and green alternately. I did multiple sets of tests:
a. Divide cf1 and cf2 into one channel;
b. Divide cf1 and cf2 into two channels and use two PAs;
c. I also tried to replace the flashing cf2 with a new one;
d. I also tried to replace the flashing cf2 with a new mark;
But the final result is that cf1 flight is very stable, but cf2 will drift to a certain extent or ignore the command and do not fly.

Have you experienced a similar situation?At the same time, I would like to ask, if single marker point tracking is performed, what is the size of the marker point generally selected? @whoenig  @jpreiss

whoenig, 2020-09-06:
Sorry for the delay - I was on vacation with limited Internet access.

**Latency**: I think 20ms is OK - I have seen such numbers on other machines without any issues flying as well. For the measurement, the PC sends a message with a timestamp to the CF, the CF sends it back, and then the latency is computed as (arrival time - sent time) / 2. Higher latency can depend on many settings, including which USB port you use to attach your Crazyradio to and what decks you have attached to your CF.

**Tracking**: In the single-marker case the attitude is computed using the on-board EKF. This is very robust as long as you don't hover for > 30s in one place (i.e., the EKF requires you to move in the xy-plane to recover the attitude correctly). In the four-marker case, the attitude is directly measured by your mocap. If your mocap would operate perfectly, four-markers are better. However, in many practical cases the coverage of the Mocap is not perfect, or the markers are too small. If you have temporary occlusions, the single-marker mode can be more robust in recovering a drone. I personally almost always use the single-marker mode these days. The "correct" marker size depends on your mocap (camera resolution + distance). You should switch to grayscale mode in Motive and count the number of pixel that a marker takes in the camera frame - at least 5 pixels diameter are recommended for Optitrack.

**Firmware**: I've seen the `OW: Cmd 0x21 timeout error` as well and it should not affect correct flight operation.

**Flight Issues**: I assume you can fly cf1 well independently and cf2 also well independently, but the issues do occur when flying both at the same time? I suggest you repeat your earlier tests: Do not fly, but move around both drones manually with logging enabled and make sure that the result is as expected. From your description, it sounds like cf2 either does not get any mocap update, or it receives the mocap update for cf1 instead. Also make sure that your NRF51 firmware is up-to-date.

swarm5, 2020-09-07:
Thank you very much for your suggestion and wish you a happy holiday.

closed 2020-09-07 by swarm5