USC-ACTLab/crazyswarm #248: Is ros-vrpn dependency in CI and installation docs necessary?

jpreiss, 2020-09-04:
In our installation instructions (https://github.com/USC-ACTLab/crazyswarm/blob/master/docs/installation.rst#simulation-and-physical-robots) we tell the user to install `ros-kinetic-vrpn-client-ros`.

In our CI script

https://github.com/USC-ACTLab/crazyswarm/blob/dfd69b44fcb8fe78f11fa3d76e1a35e4cea715c8/.github/workflows/ci-ros.yml#L34 
we install the package `ros-<distro>-vrpn`.

@whoenig some questions:

1. Why are these different?
2. Are these really necessary? I thought VRPN was optional just like vicon, optitrack, etc.

whoenig, 2020-09-06:
They are different packages and all are optional. If people use it, they should use http://wiki.ros.org/vrpn_client_ros, which is newer.

jpreiss, 2020-09-09:
@whoenig it looks like [vrpn_client_ros](http://wiki.ros.org/vrpn_client_ros) is not available on Noetic yet. This breaks the default `libmotioncapture` build because [it enables VRPN by default](https://github.com/USC-ACTLab/libmotioncapture/blob/d523b08d151c7d951c57a0b8d53062a2d82dcad1/CMakeLists.txt#L10). Any reason we cannot disable this in `libmotioncapture` and remove VRPN from our CI and instructions?

whoenig, 2020-09-09:
We can simply disable it. I don't find VRPN that useful for large swarms since it only supports object tracking (and can not provide a pointcloud). Technically, we don't rely on the vrpn_client_ros, but on libvrpn. So if we really wanted to keep it, we could also add https://github.com/vrpn/vrpn as a submodule.

jpreiss, 2020-09-09:
I have not seen VRPN mentioned often in user issues, so making it optional seems fine.

jpreiss, 2020-09-09:
If I understand correctly, compiling `libmotioncapture` currently will only work if the `VRPN` package has been built with CMake one way or another, e.g. through the ROS package or by compiling `libvrpn` manually. Should we document this in the `libmotioncapture` readme and then point to it from the `crazyswarm` docs?

whoenig, 2020-09-10:
I don't think it matters if it was build with CMake or not: CMake just needs to be able to find the header files and library.

jpreiss, 2020-09-28:
VRPN disabled by default in https://github.com/USC-ACTLab/libmotioncapture/pull/9

closed 2020-09-28 by jpreiss