USC-ACTLab/crazyswarm #249: Hovering is unstable

864419409, 2020-09-04:
Hello, I would like to ask, why in the process of many aircraft hovering or flying, there will be a situation of aircraft rotation, or even falling, how to improve, thank you

whoenig, 2020-09-06:
There is some information missing. Which tracking mode are you using? How many CFs? What kind of issues are you seeing exactly?

864419409, 2020-09-07:
hello whoening ，The tracking mode I chose is defaultSingleMarker .I chose eight planes at the same time.But when I execute the trajectory（niceHover.py），the plane might spin or fall in about 10 seconds.
I also use some simple flight paths, which can happen even if the plane can't execute the path (stay in place). But the plane is actually connected to the switch

whoenig, 2020-09-09:
Does flying a single CF work?

closed 2021-02-20 by jpreiss