USC-ACTLab/crazyswarm #252: How to recommend channels?

swarm5, 2020-09-07:
I am trying to find the best indoor channel recently. I used `https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio-test.py.`
I don't quite understand the recommended rules for the best channel. The following figure is the result of one run.
**In my understanding, the ack rate and RSSI average, the higher the better.** Obviously, 70 channels should be significantly better than other channels. But the rule in the program seems to be looking for ACK=1, and RSSI does not have such a high channel. Is there a problem with the recommendation algorithm?
[image]

whoenig, 2020-09-09:
Ack is more important than RSSI. A high RSSI can also happen by adverse effects from your environment. Your plot looks great throughout - any channel should do. Just make sure you space different channels at least 2 apart, because in the 2M mode consecutive channels are overlapping. The recommendation algorithm is pretty simplistic and should not be taken too seriously.

swarm5, 2020-09-09:
This is consistent with my guess, I have no other doubts, thank you. I will close this topic.

closed 2020-09-09 by swarm5