USC-ACTLab/crazyswarm #253: mocap_helper gives occluded subject, while visible in Nexus

kmerckae, 2020-09-10:
Hello,

I created a _cf1_ subject in Nexus 2.7.0 and launched `roslaunch crazyswarm mocap_helper.launch`.

When I don't select my created subject in Nexus and so see the separate markers in Nexus, I receive the position of the separate markers. 
[image]

When I select my created subject _cf1_ in Nexus (including 4 markers), I can see the subject with the connection lines in Nexus, but receive from the mocap_helper.launch that _cf1_ is occluded. Though I still receive the position of the marker that is not part of _cf1_ and doesn't belong to any subject. 
[image]

Information is for sure exchanged between the Vicon computer and my computer (Ubuntu 16.04), but the moment I select the subject, my computer gets the message it is occluded while it is still visible in Nexus. 
I also tried this with the vicon_bridge of ethz-asl and got the same problem, as you can see [here](https://github.com/ethz-asl/vicon_bridge/issues/26#issue-697492338).  

Does anyone know what can cause this problem? 

Thanks in advance!
Kelly

whoenig, 2020-09-11:
Both vicon_bridge and the crazyswarm use the official VICON DataStream SDK. The latest version (https://www.vicon.com/software/datastream-sdk/) comes with Python support and should be pretty easy to test. I suggest that you try to reproduce the issue using the SDK. If it is reproducible, it is best to reach out to VICON support for help. If the latest SDK works, we can try to upgrade the crazyswarm to the latest SDK (we use 1.7.0 right now) and see if the behavior is gone. Sorry, we don't have Nexus, so we won't be able to reproduce on our side.

kmerckae, 2020-10-09:
Hello, 

We have tried the latest Vicon DataStream SDK via the ViconDataStreamSDK_CPPTest. This was only giving us marker positions, but no object positions nor orientations. So we had the same problem as with the mocap_helper.launch from crazyswarm. 

Apparently in Nexus you have to make sure that the processing output level is **kinematic fit**, like in the screenshot below. 
[image]

By changing this to kinematic fit, we also obtained the positions and orientations of objects when running the ViconDataStreamSDK_CPPTest. Both the mocap_helper.launch and the vicon_bridge are working as they should right now!

closed 2020-10-09 by kmerckae

whoenig, 2020-10-10:
Thanks for the update!