USC-ACTLab/crazyswarm #254: No module named 'crazyflie'

ThomasCarstens, 2020-09-10:
Hello!
I have recently set up crazyswarm on the lab's Ubuntu 18.04.
The ./build.sh script has compiled, but the python scripts give a bug:
"ModuleNotFoundError: No module named 'crazyflie'"

I followed the crazyswarm installation instructions, and am compiling on a conda environment. I see the crazyswarm.py package wants to import the crazyflie module, should I be importing pycrazyswarm.crazyflie? Perhaps I missed something.

ThomasCarstens, 2020-09-11:
The cfclient gui connects to the drone, leading me to think perhaps this is an issue concerning crazyswarm.

ThomasCarstens, 2020-09-11:
Hello, having uninstalled and reinstalled the framework, I still face the same problem. It seems that 'crazyflie' is a catkin metapackage but it is not recognized in python3 (nor python2.7, where there is a segfault and close inspection shows the same crazyflie package not found). 
I can upload my ./build.sh script for now.
```
(crazyswarm) dvic@feuuu:~/crazyswarm$ ./build.sh
Submodule path 'externalDependencies/qualisys_cpp_sdk': checked out 'eac6b67598802c5a66c8095530010a5c6835db11'
Submodule path 'externalDependencies/vicon-datastream-sdk': checked out '94aa04411d6d03a7d13ec2832673f6f660e70616'
arm-none-eabi-size cf2_nrf.elf
   text	   data	    bss	    dec	    hex	filename
  26428	    444	   2512	  29384	   72c8	cf2_nrf.elf
S110 Activated
BLE  Disabled
Built for platform cf2
  CLEAN_VERSION
  VTMPL version.c
  CC    version.o
  LD    cf2.elf
  COPY  cf2.hex
  COPY  cf2.bin
  DFUse cf2.dfu
Build for the CF2 platform!
Build 8:febc78015a9b (2019.09 +8) CLEAN
Version extracted from git
Crazyloader build!
   text	   data	    bss	    dec	    hex	filename
 188752	   3904	  98000	 290656	  46f60	cf2.elf
Crazyradio PA build
python2 setup.py build_ext --inplace
running build_ext
Base path: /home/dvic/crazyswarm/ros_ws
Source space: /home/dvic/crazyswarm/ros_ws/src
Build space: /home/dvic/crazyswarm/ros_ws/build
Devel space: /home/dvic/crazyswarm/ros_ws/devel
Install space: /home/dvic/crazyswarm/ros_ws/install
####
#### Running command: "cmake /home/dvic/crazyswarm/ros_ws/src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=/home/dvic/crazyswarm/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/dvic/crazyswarm/ros_ws/install -G Unix Makefiles" in "/home/dvic/crazyswarm/ros_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/dvic/crazyswarm/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/dvic/crazyswarm/ros_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/dvic/crazyswarm/ros_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /home/dvic/anaconda3/envs/crazyswarm/bin/python2 (found suitable version "2.7.18", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /home/dvic/anaconda3/envs/crazyswarm/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dvic/crazyswarm/ros_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /home/dvic/anaconda3/envs/crazyswarm/bin/python2 (found version "2.7.18") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 10 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - libmotioncapture
-- ~~  - libobjecttracker
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_controller
-- ~~  - crazyflie_driver
-- ~~  - crazyswarm
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
-- ==> add_subdirectory(crazyflie_ros/crazyflie)
-- +++ processing catkin package: 'crazyflie_cpp'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_cpp)
[... 676 lines cut ...]
[ 18%] Built target console
[ 20%] Built target listMemories
[ 20%] Built target scan
[ 21%] Built target listParams
[ 23%] Built target version
[ 24%] Built target listLogVariables
[ 25%] Built target flash
[ 26%] Built target comCheck
[ 27%] Built target setParam
[ 30%] Built target log
[ 30%] Built target downloadUSDLogfile
[ 32%] Built target ViconDataStreamSDK_CPPTest
[ 44%] Built target crazyflie_driver_generate_messages_py
[ 55%] Built target crazyflie_driver_generate_messages_cpp
[ 68%] Built target crazyflie_driver_generate_messages_eus
[ 78%] Built target crazyflie_driver_generate_messages_lisp
[ 82%] Built target libmotioncapture
[ 94%] Built target crazyflie_driver_generate_messages_nodejs
[ 95%] Built target crazyflie_server
[ 96%] Built target crazyflie_add
[ 96%] Built target crazyflie_driver_generate_messages
[ 97%] Built target crazyswarm_teleop
[ 98%] Built target mocap_helper
[100%] Built target crazyswarm_server
```

whoenig, 2020-09-12:
The missing Python package has nothing to do with cflib/cfclient and is rather the ROS package. Are you using the same Python interpreter that catkin has found (`/home/dvic/anaconda3/envs/crazyswarm/bin/python2` from the logs) when you run a test script?

ThomasCarstens, 2020-09-12:
Yes, python version is 2.7 within the conda environment. 
```
$ python individual_hover.py 
Segmentation fault (core dumped)
```
Not sure what piece of memory is causing the core dump, but according to the stacktrace there are files I'm not authorised in writing to. Probably python2.7 configuration in the virtualenv. Here is the last part of the stacktrace (as it is very long):
```
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/numpymodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/numpy.py", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/numpy.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
stat("/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/PyKDL", 0x7ffe0b53f1c0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/PyKDL.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/PyKDLmodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/PyKDL.py", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/tf_conversions/PyKDL.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
stat("/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/PyKDL", 0x7ffe0b53f1c0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/PyKDL.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/PyKDLmodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/PyKDL.py", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/PyKDL.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
stat("/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/PyKDL", 0x7ffe0b53f1c0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/PyKDL.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/PyKDLmodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/PyKDL.py", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/PyKDL.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
stat("/opt/ros/melodic/lib/python2.7/dist-packages/PyKDL", 0x7ffe0b53f1c0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/PyKDL.so", O_RDONLY) = 6
fstat(6, {st_mode=S_IFREG|0644, st_size=414320, ...}) = 0
openat(AT_FDCWD, "/opt/ros/melodic/lib/python2.7/dist-packages/PyKDL.so", O_RDONLY|O_CLOEXEC) = 8
read(8, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\240f\1\0\0\0\0\0"..., 832) = 832
fstat(8, {st_mode=S_IFREG|0644, st_size=414320, ...}) = 0
mmap(NULL, 2509384, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 8, 0) = 0x7f1fe0479000
mprotect(0x7f1fe04d1000, 2097152, PROT_NONE) = 0
mmap(0x7f1fe06d1000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 8, 0x58000) = 0x7f1fe06d1000
close(8)                                = 0
openat(AT_FDCWD, "/home/dvic/anaconda3/envs/crazyswarm/bin/../lib/liborocos-kdl.so.1.4", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/liborocos-kdl.so.1.4", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/melodic/lib/liborocos-kdl.so.1.4", O_RDONLY|O_CLOEXEC) = 8
read(8, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\20\24\2\0\0\0\0\0"..., 832) = 832
fstat(8, {st_mode=S_IFREG|0644, st_size=611376, ...}) = 0
mmap(NULL, 2706584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 8, 0) = 0x7f1fe01e4000
mprotect(0x7f1fe0276000, 2097152, PROT_NONE) = 0
mmap(0x7f1fe0476000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 8, 0x92000) = 0x7f1fe0476000
close(8)                                = 0
mprotect(0x7f1fe0476000, 8192, PROT_READ) = 0
mprotect(0x7f1fe06d1000, 4096, PROT_READ) = 0
stat("/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/sip", 0x7ffe0b53ee20) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/sip.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/sipmodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/sip.py", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/src/crazyswarm/scripts/sip.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
stat("/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/sip", 0x7ffe0b53ee20) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/dvic/crazyswarm/ros_ws/devel/lib/python2.7/dist-packages/sip.so", O_RDONLY) = -1 ENOENT (No such file or directory)
[... cut ...]
mmap(NULL, 2223584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 9, 0) = 0x7f1fdffc5000
mprotect(0x7f1fdffe2000, 2093056, PROT_NONE) = 0
mmap(0x7f1fe01e1000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 9, 0x1c000) = 0x7f1fe01e1000
close(9)                                = 0
mprotect(0x7f1fe01e1000, 4096, PROT_READ) = 0
close(8)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x720} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)
```

jpreiss, 2020-09-17:
It seems like before when you got the import error, you were running a Python 3 interpreter. Python 3 changed the way relative imports work, causing this "no module named `crazyflie`" error.

I am not sure about the segfault. To ensure that the firmware bindings are compiled for the same version of Python that you are now running, you can run `make clean; make` in the `pycrazyswarm/cfsim` directory.

@whoenig have you ever run with ROS and a non-system Python interpreter? I haven't. Our ROS setup instructions assume that you will use the system Python interpreter. We install some Python packages via `apt` instead of `pip`. @ThomasCarstens please try following our ROS setup documentation with the system Python to see if the problem persists.

whoenig, 2020-09-18:
I never tried any non-system Python.

ThomasCarstens, 2020-10-19:
Alright, with the system Python, the problem is solved. I have no need to use anaconda for now

closed 2020-10-19 by ThomasCarstens