USC-ACTLab/crazyswarm #258: Crazyswarm documentation

MauroPfister, 2020-09-24:


jpreiss, 2020-09-25:
You are totally right about the `crazyswarm_server.cpp` issue. I believe the problem is that the Crazyswarm version links directly against https://github.com/USC-ACTLab/libobjecttracker (providing the feature of tracking multiple objects with the same marker configuration), whereas the `crazyflie_ros` version does not. It would be nice if we didn't have to have two versions of the file, but I guess that would require some conditional compilation setup in `crazyflie_ros`. @whoenig I recall discussing this with you, but cant remember if it's a WONTFIX or not.

For the yaw arguments, I'm sure you are right but I would like to double check first and add code comments for the appropriate functions/structs - somehow there was wrong/missing info in the code that  caused the documentation error.

Pull requests would be great! Please make separate commits for each change (or separate PRs). For small and "uncontroversial" changes in the future, feel free to create a PR directly without discussing in an issue first.

whoenig, 2020-09-25:
Longer-term (perhaps early next year), the goal is to merge the two repositories. In my opinion, there is very little use of crazyflie_ros and I almost always use the crazyswarm instead (even when flying a single CF). However, crazyflie_ros still has additional features that we didn't port yet and is a bit easier to debug because it does not use multithreading.

+1 for fixing the docs!

MauroPfister, 2020-09-26:
Thanks for the clarification about `crazyswarm_server.cpp`. 

Shall I make a PR for the yaw arguments or will you fix this yourself after testing @jpreiss ? Concerning the tutorial about creating a new streaming setpoint mode: Maybe it is better if someone with a deeper knowledge of the cpp code fixes that part.