USC-ACTLab/crazyswarm #259: ESTKALMAN: State out of bounds, resetting

swarm5, 2020-10-04:
@whoenig @jpreiss **I use the latest branch. But in the process of operation, every cf will have the same prompt, I don't know if this will affect the flight experiment**.` [INFO] [1601817211.120082235]: [cf8] ESTKALMAN: State out of bounds, resetting`

In the experiment, cfs can fly according to my predetermined trajectory, **but sometimes some planes are not very stable, they will fly around the hovering point while flying**. (I will select a few videos to upload later)
I use a radio to control 9 cfs. I use a single marker in libobjecttracker mode. At the same time, I use a 19.5mm marker.
My experiment site is about 5×6×3, and the system has 12 17Wprimes in total, and each lens has 170W pixels.
**I don’t know where I have the problem now?**

swarm5, 2020-10-04:
`$ roslaunch crazyswarm hover_swarm.launch`

> ... logging to /home/ss/.ros/log/6553f2aa-0643-11eb-ae95-b42e99612c7c/roslaunch-407-12867.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> WARNING: disk usage in log directory [/home/ss/.ros/log] is over 1GB.
> It's recommended that you use the 'rosclean' command.
> 
> started roslaunch server http://localhost:41189/
> 
> SUMMARY
> 
> PARAMETERS
>  * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
>  * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
>  * /crazyflieTypes/CF21SingleMarker/bigQuad: False
>  * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
>  * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
>  * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
>  * /crazyflieTypes/default/batteryVoltageWarning: 3.8
>  * /crazyflieTypes/default/batteryVoltateCritical: 3.7
>  * /crazyflieTypes/default/bigQuad: False
>  * /crazyflieTypes/default/dynamicsConfiguration: 0
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
>  * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
>  * /crazyflieTypes/default/markerConfiguration: 3
>  * /crazyflieTypes/defaultSingleMarker/batteryVoltageWarning: 3.8
>  * /crazyflieTypes/defaultSingleMarker/batteryVoltateCritical: 3.7
>  * /crazyflieTypes/defaultSingleMarker/bigQuad: False
>  * /crazyflieTypes/defaultSingleMarker/dynamicsConfiguration: 0
>  * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
>  * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
>  * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
>  * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
>  * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kR_z: 60000
[... 747 lines cut ...]
> [ INFO] [1601817271.990741796]: [cf17] Uploaded trajectory
> [ WARN] [1601817271.993196399]: Latency too high! Is 0.083587 s.
> [ INFO] [1601817271.999635433]: Takeoff!
> [ INFO] [1601817274.538832496]: [cf8] GoTo
> [ INFO] [1601817274.551328056]: [cf10] GoTo
> [ INFO] [1601817274.563326599]: [cf11] GoTo
> [ INFO] [1601817274.576235496]: [cf12] GoTo
> [ INFO] [1601817274.588289374]: [cf13] GoTo
> [ INFO] [1601817274.601379895]: [cf14] GoTo
> [ INFO] [1601817274.613326436]: [cf15] GoTo
> [ INFO] [1601817274.626192240]: [cf16] GoTo
> [ INFO] [1601817274.638007471]: [cf17] GoTo
> [ INFO] [1601817277.152737287]: Start trajectory!
> [ INFO] [1601817286.484493668]: Start trajectory!
> [ INFO] [1601817295.806028606]: Land!
> ^C[rviz-5] killing on exit
> [crazyswarm_teleop-4] killing on exit
> [joy-3] killing on exit
> [crazyswarm_server-2] killing on exit
> [rosout-1] killing on exit
> [master] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
>

swarm5, 2020-10-04:
@whoenig @jpreiss  I ran nicehover.py and this video is the result of my flight experiment: https://youtu.be/h7SdBY4S39k.
There are two problems:
1. Obviously, cf starts to circle soon after takeoff, which greatly affects the stability of the flight.
2. There is another problem that has been bothering me. Look carefully at the second line of the video, the left and the right CFS, after flying, their led ring starts to flicker. Similarly, when it landed, they still flickered.

whoenig, 2020-10-04:
1. It looks weirdly badly tuned, like too high D-gains. From the log it seems like you might be using the wrong cftype. Since you seem to have CF 2.1, make sure you use `CF21SingleMarker` as your type in `allCrazyflies.yaml`. Also, according to the `ESTKALMAN` error, it might be that your EKF does not behave as expected. Do you have uSD cards and decks by any chance? This way you could log the raw data and state estimate and make sure that the state estimate is accurate. Finally, do you observe similar behavior when you fly fewer CFs? If so, does the behavior just depend on the number of CFs you fly, or are specific ones causing the circular oscillations?

2. I think you refer to the flashing of the green light? In the default LED ring effect, this just means that the battery is low (or was low during the flight). If your batteries are old, this can be triggered very quickly, even shortly after takeoff. For new batteries, you should be able to fly for a couple of minutes before the flashing starts.

jpreiss, 2020-10-04:
Did you update the firmware on the CF too?

swarm5, 2020-10-05:
> We recently updated the firmware version linked by Crazyswarm. Are you using the new firmware?

Yes. What I want to say is that when I chose the master branch, I had already updated two firmware using chooser.py. At the same time, I also updated the firmware of the radio using the ros command.

swarm5, 2020-10-05:
@whoenig 
Thank you very much for your good suggestions, I will test it tomorrow.
1. cftype: I did choose cftype by mistake. I have been using defaultSingleMarker. I didn't notice CF21SingleMarker before. The cf I use is version 2.1. I want to ask, what kind of situation is the defaultSingleMarker type suitable for?
2. usd: I only have one usd card, and I am purchasing more. However, I don't know how to use it with crazyswarm? For example, how to make the usd card record the log during the flight? Where should I go to check the user guide of the usd card?
3. led ring: led ring flashes, I already understand the reason, I will replace the battery in time, thank you!

whoenig, 2020-10-06:
1. The name is misleading for historical reasons. `defaultSingleMarker` should be `CF20SingleMarker`, i.e., it applies for a regular Crazyflie 2.0 with a single marker on top.

2. There is no good documentation from our side yet. Basically, you put a config.txt on the uSD card with a list of variables you want to log into a binary file. After the flight, the file can be downloaded over the radio or by inserting the uSD card into a card reader on your PC. Finally, you can decode the file and plot the data. Some (limited) information is available at https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:microsd#data_logging. I usually look at the stateEstimate.{x,y,z} variables at the very least.

swarm5, 2020-10-09:
@whoenig I used two cfs to do the same experimental test:
1. When I use the first cf to run niceHover.py, the cf rolls over immediately. I have done the relevant experiments four times, and each time it rolls over immediately. But I looked at the data of stateEstimate.{x,y,z}, there should be no problem.
2. I changed to a cf and used the same method to run niceHover.py. This time cf just didn't fly very well.

I have some questions:
1. How to record the raw data of optitrack? I tried optitrack.x and ext_pose.x, but the names of these two variables seem to be wrong.
2. Some cf seems to be a problem, I guess it may be a parameter problem. I used the remote control to fly the first plane, and there was no rollover. Have you encountered similar problems before?

swarm5, 2020-10-09:
[image]
@whoenig @jpreiss 
This picture is a typical record of the problems I encountered, cf will produce oscillation or circle phenomenon.
Let me explain. This picture records three flights.
1. For the first time, I ran niceHover.py and the flight was very stable.
2. The second time, immediately after I ran figure8.py, there was a slight oscillation and circle in the second half.
3. For the third time, I did it again, and figure8.py was also running. This time it flew very unstable and fell to the ground.

swarm5, 2020-10-09:
I conducted another experiment, and I made the results public for your reference:
Experiment procedure:
1. I fly 8 cfs at the same time, but I only have one usd card, and I only record one of them.
2. I flew three times, the first and third times were niceHover, and the second time was figure8.py.
3. It can be seen from the data that the third flight has already produced severe oscillations.

Doubt:
1. I used ext_pos to successfully record OptiTrack data, but the result is very strange. Figure 1 is the data recorded in the log, and Figure 2 is an enlarged ext_pos data. **I found that ext_pos data is similar to many aperiodic triangular pulse signals. I don’t know what the problem is? ? ?** (My log data and processing program can be downloaded here:)
**2. Who can analyze from the log data the reason for the oscillation during the flight?**
[image]
[image]

whoenig, 2020-10-10:
Thanks for the detailed report. I think for now it is best to focus on the single CF that consistently rolls over. It is worrisome that that your ext_pos.{X,Y,Z} data is often 0. There seems to be a [bug in the firmware](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_localization_service.c#L208-L218) that makes the data incorrect if you fly multiple CFs (to fix it, move lines 208-211 inside the if condition before line 213 -or- just fly a single CF). While this bug might explain the triangular signal you are seeing, it does not explain why you got 0 in your earlier experiments. Did you update the NRF51 as well as the Crazyradio firmwares also?

Rollover and oscillations are often associated with issues related to the attitude. You could use the following config file to collect some more information:
```
100   # frequency
20    # buffer size
log   # file name
0     # enable on startup (0/1)
1     # mode (0: disabled, 1: synchronous stabilizer, 2: asynchronous)
stateEstimateZ.x
stateEstimateZ.y
stateEstimateZ.z
stateEstimateZ.vx
stateEstimateZ.vy
stateEstimateZ.vz
stateEstimateZ.ax
stateEstimateZ.ay
stateEstimateZ.az
stateEstimateZ.quat
ctrltargetZ.x
ctrltargetZ.y
ctrltargetZ.z
ctrltargetZ.vx
ctrltargetZ.vy
ctrltargetZ.vz
ext_pos.X
ext_pos.Y
ext_pos.Z
```
Ideally, you only start/stop logging in you script (hover or figure8) at the beginning/end by setting the `usd.logging` to 1 and 0, respectively.

swarm5, 2020-10-12:
@whoenig Thank you very much for your answer. I was on a business trip a few days ago and there was no time for further testing. Now I am back, I will conduct further testing tomorrow morning (9 hours later).
I have a few questions, I need to confirm with you:
1. If I have modified the bug in the firmware, then I need to run ./build.sh to compile, and finally use chooser.py to upload the firmware to cf?
2. Set enable on startup to 0, which is a better choice, but I am not sure if I fully understand what you mean. I need to add ‘usd.logging=1’ after importing the module in niceHover.py, and then add ‘usd.logging=0’ at the end of the program?

whoenig, 2020-10-13:
1. Yes. You can also just type `make` in the firmware folder to compile (but `./build.sh` will do that for you also).
2. At the beginning of the script, add `allcfs.setParam("usd/logging", 1)` and at the end add `allcfs.setParam("usd/logging", 0)`.

Good luck with your tests!

swarm5, 2020-10-13:
I just added the test and I will publish the results:
### 1.Single flight
**1.1 Experimental procedure**
Number: 1
Mode: CF21SingleMarker
Program: niceHover-->figure_8-->niceHover
**1.2 Results**
**The log data of ext_pos is still 0**, and the firmware may have a bug.
**The flight effect is consistent with my previous test.** During the first niceHover, the flight was stable, but during the process of figure_8-->niceHover, the aircraft began to oscillate and circle.
**1.3 Experimental diagram**

- Figure 1: ext_pos data, the results are the same three times

[image]

- Figure 2: The first flight, niceHover

[image]

- Figure 3: The second flight, figure_8

[image]

- Figure 4: The third flight, niceHover

[image]

### 2.Multiple cfs flights
I am experimenting, I will update soon

swarm5, 2020-10-14:
I want to update my situation:
Today, I tried to record the value of ext_pos.X with crazyswarm back in 2019. Unfortunately, its value is still 0.
[image]

swarm5, 2020-10-15:
I think I should summarize the results of recent experiments to provide reference for others:
1. [cf8] ESTKALMAN: State out of bounds, resetting, this prompt will not lead to bad results. This seems to be a very normal prompt, cf often flies very well when there is this prompt.
2. The oscillation phenomenon of cf seems to be related to cf itself. I used the same firmware and settings to fly 18 cfs, and found that there were always so few that produced oscillations. Whether I fly a single cf or multiple cfs, the 7 cfs are always the ones that oscillate.
**I don't know, is this because these cfs use worse sensors, or do you need to replace the motors and propellers? If anyone has experience with this, please let me know! ! ! I will be very grateful.**
3. Regarding the ext_pos data log, I found that this is also related to cf itself. Some cfs, whether it is a single flight or multiple flight
,. However, for the other part of cfs, the data of ext_pos in single-plane flight is 0, but the data of ext_pos in multi-plane flight is normal. I don't know the reason? This seems to be related to the hardware. I use cf2.1, but I bought it in two batches. From the results of my experiment, the data of ext_pos being 0 does not affect the normal flight of cf.

swarm5, 2020-10-15:
I will close this issue in three days, because I think the prompt `"ESTKALMAN: State out of bounds, resetting"` has no practical impact on my flight.
I will continue to discuss the following issues in the new issue:
1. The problem of vibration in flight (this seems to be related to cf hardware)
2. The problem that ext_pos data is recorded as 0 (this seems to be related to the cf hardware, but it has no direct effect on the flight results)

whoenig, 2020-10-16:
1. Oscillations: Try using the PID controller rather than Mellinger. I have seen these kinds of issues before and they usually mean that the controller gains are tuned too aggressively (i.e., gains are too high). You are right that your 7 CFs probably have worse propellers, motors, or noisier sensors, but ultimately for CF2.0 we were able to find gains that work over a wide range of CFs. For CF 2.1 we just have fewer units, so it is not as well tuned.

2. Your stateEstimate data looks good in the plots, so I assume this is just some issue with the logging. Can you confirm that you are still using just a single marker for tracking? Because `ext_pos` is not updated in the firmware for the use-case of full-pose tracking (4 markers).

swarm5, 2020-10-16:
1. I have several sub-questions:
**a)** What is the difference between PID and Mellinger controllers? I will use cf for trajectory flight and cmdVelocityWorld command. Is it better for me to use the Mellinger controller in the trajectory experiment; and in the cmdVelocityWorld experiment, I use the PID controller?
**b)** Does Mellinger need no parameters? In the crazyflieTypes.yaml file, I only saw the parameters of the PID controller? I measured the weight of my cf, it is about 37g, do I need to adjust the parameters?
```
      ctrlMel:
        kp_xy: 0.4 #0.6 # p for position
        kd_xy: 0.2 # 0.35 # d for position
        ki_xy: 0.05
        i_range_xy: 2.0
        kR_xy: 70000 # p for attitude
        kw_xy: 20000 # d for attitude
        kR_z: 60000 # 70000 # p for yaw
        kw_z: 12000 # 30000 # d for yaw
        ki_m_z: 500 #0 #20000 # i for yaw
        i_range_m_z: 1500 # i range for yaw
        kd_omega_rp: 200 # roll and pitch angular velocity d gain
        kp_z: 1.25
        kd_z: 0.4
        ki_z: 0.05
        i_range_z: 0.4
        mass: 0.032
        massThrust: 132000
```
**c)** I have observed that the difference between defaultSingleMarker and CF21SingleMarker is only the Kalman parameter. At the same time, as far as I know, the difference between CF2.1 and CF2.0 is that it only replaces a sensor. Does the Kalman parameter have to oscillate in flight?
```
kalman:
        pNAcc_xy: 1.0 # default 0.5
        pNAcc_z: 2.0 # default 1.0
        mNGyro_rollpitch: 0.1 # default 0.1
        mNGyro_yaw: 0.1 # default 0.1
```
**d)** Now, I can only buy CF2.1, so I found a set of parameters that are widely applicable to CF2.1. How can I adjust these parameters according to the flight situation? Is there any information for reference?

2. Yes, I can confirm that I still use a single marker for tracking.

swarm5, 2020-10-19:
I still haven't completely solved the cf oscillation problem. If I can find a new method in the future, I will open an issue to discuss it.

closed 2020-10-19 by swarm5