USC-ACTLab/crazyswarm #263: interactivity demo

swarm5, 2020-10-19:
Hello, I watched the promotional video of crazyswarm a large nano-quadcopter swarm the other day. Among them, **the interactivity part** is very attractive to me. I want to find a demo to try.
I found two files avoid_human_demo.m and avoidtarget_mex.c in the matlab folder of iros2018, but I don't understand how to use them? Do you have any good suggestions? @jpreiss @whoenig

swarm5, 2020-10-19:
If I want to set the T-shaped calibration rod as an object that needs to be avoided, how can I feed back the position of the T-shaped calibration rod to cfs in real time in crazyswarm?

jpreiss, 2020-10-19:
That functionality was removed when we switched from a forked firmware to the official firmware. There were special cases in the object tracker and firmware to send the position of the calibration wand. If I recall correctly, we dropped the feature because it was a little overly specific to that one demo.

Maybe the easiest way to recreate it is to wait for bitcraze to accept  https://github.com/bitcraze/crazyflie-firmware/pull/628 . Then you can take off all but one CF, enable collision avoidance, and carry around the last CF instead of the calibration wand. It's a totally different algorithm though, so the behavior might look a little different.

swarm5, 2020-10-19:
@jpreiss Thank you for your answer. However, in fact, I prefer to create a rigid body and send the position of the rigid body to cfs in real time. Can this be done?
Because I want to try the game scene, similar to the eagle rushing to disperse the birds. I want to use the Xbox handle to control a cf (this cf has been established as a rigid body) to achieve interaction with the cf group.

whoenig, 2020-10-20:
It looks like the crazyswarm server itself still has this support, at least partially (see e.g., https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L879-L882). If a the virtual object (currently only supported with VICON, but it should be easy to add for Optitrack or others) is visible, it will be broadcasted as ID 255. The firmware portion was, as James wrote earlier, removed when we switched to the official firmware.

jpreiss, 2020-10-20:
The new collision avoidance algorithm should also work if you are controlling one of the Crazyflies manually with a controller.

swarm5, 2021-02-20:
@jpreiss hi, I am concerned that the Collision avoidance function you developed has passed.
I am trying to fly 20 cfs to simulate some mission scenarios. The overall process is that my senior planner on the PC side can calculate the expected position of each cf. Because there are 20 airplanes in the scene, the collision avoidance problem between cfs needs to be considered when going to the desired location.
I have a few questions I would like to ask you:
1. I think you stated in bitcraze/crazyflie-firmware#628  that each drone can only receive the location information of other cfs in the same radio channel, so does it indicate that this function can only achieve collision avoidance on the same channel cf? Obviously, my 20 cfs cannot be set up under the same radio channel. In this case, what do you recommend?
2. I want to test the multi-cfs collision avoidance function, how do I start? I saw that you mentioned in your answer that the bvca branch of crazyswarm has integrated the multi-cf collision avoidance function. I would like to ask whether this branch has been tested and matured now? Has the function of collision avoidance been completed? Whether the bvca branch can achieve my goal (I hope to achieve 20 cfs flying to the destination while avoiding collisions).

swarm5, 2021-02-20:
> If a the virtual object (currently only supported with VICON, but it should be easy to add for Optitrack or others) is visible, it will be broadcasted as ID 255.

Thanks for your answer, but I don’t quite understand the virtual object. Is this a feature in vicon? I wonder if you can explain it in detail?

jpreiss, 2021-02-20:
Using onboard collision avoidance with 20 CFs is not possible right now due to the multithreading architecture of the Crazyswarm server:

https://github.com/USC-ACTLab/crazyswarm/blob/e2229d5569bfdfd323fefbeba21a10cd7617df8e/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L73-L84

To broadcast the positions of all Crazyflies on all channels, we would need to add a synchronization point where each thread shares the results of its object tracking with the other threads. This is a nontrivial feature and will need to be carefully tested for performance.

`bvca` has worked for me, but only with a small number of CFs. I will be testing it more soon in the coming weeks, and merging into `master` if all goes well.

closed 2021-02-20 by jpreiss