USC-ACTLab/crazyswarm #264: Cannot use cmdVel

KeithChen-DeityShallFall, 2020-10-23:
I tried to send roll ,pitch, thrust directly to the cf, so I use cmdVel() to approach this thing. But everytime I tried to run my script it always says " AttributeError: Crazyflie instance has no attribute 'cmdVel' "  Could you help me with that?
Here's my script
#!/usr/bin/env python

import numpy as np

from pycrazyswarm import *

if __name__ == "__main__":
    swarm = Crazyswarm()
    timeHelper = swarm.timeHelper
    allcfs = swarm.allcfs
    for cf in allcfs.crazyflies:
        cf.cmdVel(0,0,0,40)
    timeHelper.sleep(3)

jpreiss, 2020-10-23:
It sounds like you are using an old version of Crazyswarm.

ThomasCarstens, 2020-10-24:
After updating the repo to the latest version, I could use it on a real drone.
Also be aware those are low level commands - more instructions for use are found in the API documentation :P

KeithChen-DeityShallFall, 2020-10-24:
Thank you! I will try to fix it! By the way, is there other method to upload thrust, pitch abd roll directly to the drone?

KeithChen-DeityShallFall, 2020-10-29:
Hi, I uploaded the repo yesterday, and the error about "no attribute" has been solved , when I use cmdVel(), I also use `rostopic echo /cf2/cmd_vel` to monitor the order. It shows that some order has published to the drone, but the drone do not move, even if I have given thrust of 2^16 to the drone. Can you show me your code about using it on a real drone? Thank you!!
[image]

whoenig, 2020-10-30:
I think the firmware has a thrust lock: the first few messages need to send all zeros, otherwise future messages are ignored.

jpreiss, 2021-02-20:
@KeithChen-DeityShallFall did sending a few messages of all zeros fix the problem?

jpreiss, 2021-03-05:
closing due to inactivity.

closed 2021-03-05 by jpreiss

adolhongping, 2021-06-11:
> Hi, I uploaded the repo yesterday, and the error about "no attribute" has been solved , when I use cmdVel(), I also use `rostopic echo /cf2/cmd_vel` to monitor the order. It shows that some order [...]

Hello, I have the same question, do you have solved it?

whoenig, 2021-06-16:
This is was closed @adolhongping. You can open a new discussion, if you still need help.

jc-bao, 2023-03-12:
I have encountered the same issue here. As Wolfgang mentioned, the problem can be resolved by sending a series of zero commands before sending the actual control command. 

BTW, you seem to send 65536, which is out of `[0, 2^16)`. Could it also be the reason?

powehi2020, 2023-03-14:
> After updating the repo to the latest version, I was able to use it on a real drone. Also be aware those are low level commands - more instructions for use are found in the API documentation :P

hi~ what version have you updated? crazyswarm? or crazyflie firmware?  do you still remenber the specific version? like 1.0.1 . thk u~

powehi2020, 2023-03-14:
> I have encountered the same issue here. As Wolfgang mentioned, the problem can be resolved by sending a series of zero commands before sending the actual control command.
> 
> BTW, you seem to send 65536, which is out of `[0, 2^16)`. Could it also be the reason?

hi~ do you use crazyswarm? but, it does not show the topic cmdVel in the topic list. am i do something wrong? or can you show your more details? thk u~

jc-bao, 2023-03-14:
You need 

> hi~ do you use crazyswarm? but, it does not show the topic cmdVel in the topic list. am i do something wrong? or can you show your more details? thk u~

Yes, I can. You are expected to see the topic after you start the crazyswarm server and any control script.

cwsung, 2024-01-30:
hi~ do you have example code of use cmdVel give linear velocity commend to the cf? I'm currently trying to use cmdVel give linear velocity commend to the cf. but even though I give series of zero commands before sending the actual control command, cf is not move at all and after loop is over, script just ended..      

here is my script

import numpy as np
from pycrazyswarm import *

swarm = Crazyswarm()
timeHelper = swarm.timeHelper
allcfs = swarm.allcfs
start = timeHelper.time()
print(start)
now = timeHelper.time()
print(now-start)
while 0 <= now - start <= 0.1:
    for cf in allcfs.crazyflies:
        cf.cmdVel(0,0,0,0)
    now = timeHelper.time()

now = timeHelper.time()
print(now-start)
while 0.1 <= now - start <= 2.1:
    for cf in allcfs.crazyflies:
        cf.cmdVel(0,0,0,60000)
    now = timeHelper.time()
print(now-start)
now = timeHelper.time()
while  2.1<= now - start <= 9:
    for cf in allcfs.crazyflies:
        cf.cmdVel(0, 5, 0, 40000)
    now = timeHelper.time()
print(now-start)