USC-ACTLab/crazyswarm #265: SetParam Issue

KeithChen-DeityShallFall, 2020-10-23:
I don't know the name of the parameters that could be set to the cfs, in the python scripting api, it said "Parameters are named values of various primitive C types that control the firmware’s behavior. For more information, see https://www.bitcraze.io/docs/crazyflie-firmware/master/logparam/." But I cannot open that website and also cannot know the name of the parameters
Thank you if anybody could tell me how to find these names of parameters!

whoenig, 2020-10-23:
You can use `rosrun crazyflie_tools listParams`, or the Bitcraze cfclient to get a list of parameter names. However, there is currently no documentation of the parameters other than the firmware source code itself.

KeithChen-DeityShallFall, 2020-10-27:
Thank you for your answer, I tried the first method:
When I tried to use` rosrun crazyflie_tools listParams`, there will be an error says "LIBUSB_ERROR_BUSY" when hover_swarm.launch is running or says "Timeout" when hover_swarm is not running.....
I'll try the other method later.

whoenig, 2020-10-30:
It needs to be executed while no other launch file is running. The timeout error indicates that you need to specify the URI of your Crazyflie using the `--uri` flag.

closed 2021-02-20 by jpreiss