USC-ACTLab/crazyswarm #266: rospy.spin() not working in real nor sim

ThomasCarstens, 2020-10-23:
I am trying to create a subscriber node.
However I cannot init_node nor use rospy.spin(), which cause problems with the CrazyflieAPI.
I do however want the subscriber to continue running, else only one pose is registered. How do I do so?
Worth noting that I am using cmdPosition() without problems.

```
#!/usr/bin/env python

import numpy as np
import rospy
import tf
from geometry_msgs.msg import TransformStamped
import tf2_msgs.msg
from pycrazyswarm import *
import uav_trajectory
import matplotlib.pyplot as plt

def callback(tf_of_interest):

    cft = tf_of_interest.transforms[0]
    print(cft)

    x = cft.transform.translation.x
    y = cft.transform.translation.y
    z = cft.transform.translation.z

    pos = np.array([x, y, z+0.3])
    print("calculated pos: ", pos)
    print ("real pos: ", allcfs.crazyflies[0].position())
    allcfs.crazyflies[0].cmdPosition(pos, 0)

def listener():
    rospy.init_node('follow_me_now', anonymous=True) # this is a maybe
    rospy.Subscriber('/tf', tf2_msgs.msg.TFMessage, callback)
    rospy.spin()

if __name__ == "__main__":
    swarm = Crazyswarm()
    timeHelper = swarm.timeHelper
    allcfs = swarm.allcfs

    Z = 0.3
    allcfs.crazyflies[0].takeoff(targetHeight=Z, duration=1.4+Z)
    timeHelper.sleep(1.5+Z)
    while (1):
        listener()

    allcfs.crazyflies[0].land(targetHeight=0.06, duration=1.0+Z)
    timeHelper.sleep(1.0)
```

ThomasCarstens, 2020-10-27:
The current 3D simulator is not ROS time-related. A ROS compatible simulator would be quite useful for myself and many others in the future. Perhaps I could get some precisions on how to change crazyflieSim.py to graph ROS data?

whoenig, 2020-10-30:
1. The Crazyflie server already initializes a node, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py#L527. Spin is not required in ROS-Python (unlike in the C++ version). Thus, you should be able to first create the swarm object and then call `rospy.Subscriber` to add additional callbacks.

2. The simulator does not rely on ROS, to reduce dependencies and allow faster simulation times. If you need ROS in this case, you'll need to call `rospy.init_node` in your script yourself (to be only active when using simulation though.) What is your use-case here? If you need more fully fledged simulation support, an integration with RotorS/CrazyS could be interesting.

closed 2021-02-20 by jpreiss