USC-ACTLab/crazyswarm #267: CMakeList.txt tracked in Git

MauroPfister, 2020-10-28:
Hi

I noticed that the autogenerated `CMakeList.txt` in `ros_ws/src` is included in the `crazyswarm` repo. This creates an issue when using a ROS version other than Kinetic and working on different branches, since the changed symbolic link results in a dirty working tree. I am not sure about best practices for ROS, but to me it seems like this file should not be tracked in Git.

Thanks for the support!

whoenig, 2020-10-30:
Good point. This should be added to `.gitignore` and not part of the repo. I think having a workspace in git is fairly uncommon, so I am not sure if there are any best practices for our particular use-case.

MauroPfister, 2020-11-27:
Hi @whoenig, are there any updates on this? Would it be possible to simply update `.gitignore` for the moment?

acxz, 2020-11-27:
I think you would also need to run `git rm ros_ws/src/CMakeList.txt` to ensure it gets removed from the repo.

PR #276 (2020-11-29): remove symlinked/autogenerated CMakeLists
closes #267
closed 2020-11-29 by jpreiss