USC-ACTLab/crazyswarm #269: Orientation not being estimated

Hurisa, 2020-10-30:
Hi everyone,

I'm experiencing an issue when publishing a simple rotation to the cmd_velocity_world topic.
That is when I start publishing, the cf actually starts rotating but after, maybe a couple of seconds, starts moving in circles and eventually crashes. I've also noticed that on RViz, the orientation doesn't seem to change even in the beginning when it appears to be rotating correctly (I say correctly, but as I start publishing the cf starts losing altitude). 
I'm guessing that it is maybe because I'm using a single marker, but when I tried the 4 smaller ones, the vicon has a hard time detecting them on the ground plane before take off.

Here are my parameters for the crazyswarm_server. 
`world_frame: "/world"

      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["pm.vbat"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 1 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          resetEstimation: 1
      # tracking
      motion_capture_type: "vicon" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      vicon_host_name: "192.168.10.1:801"
      # optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: True`

Any help is much appreciated.
Thanks

jpreiss, 2020-10-30:
You are just rotating the yaw angle while holding position? If so, there is no way it can work with a single marker. The yaw axis gyro integration will drift, and the mocap measurement cannot correct it. It is unobservable.

Hurisa, 2020-10-30:
Yeah that's what I was afraid of.
Thanks.

closed 2021-02-20 by jpreiss