USC-ACTLab/crazyswarm #271: Simultaneous linear and angular velocity

Hurisa, 2020-11-01:
Hi everyone,

Is it possible to publish messages to the cmd_velocity_world topic that are composed of a linear and angular components, such as the example below?:
```
rostopic pub /cf0/cmd_velocity_world crazyflie_driver/VelocityWorld "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'cf0'
vel:
  x: 0.40
  y: 0.0
  z: 0.0
yawRate: 30.0" -r 10
```
I am asking this because when I publish such a velocity I would expect the cf to move in a circle, i.e., moving in its `x` direction while the yaw changes. However, what I observe is the cf rotating as it drifts in one direction. Also when I publish simple velocity msgs, meaning only linear or only angular the cf moves as expected, moving in its `x` direction when the msg published  is a linear velocity, and rotating on the spot when every value is zero expect `yawRate`. I should say I have a Flow Deck.
Also I am not using the vicon, as I am having quite some trouble to get a very stable point cloud at the moment. This is indeed weird because when I was using these packages (--> https://github.com/whoenig/crazyflie_ros) I could do this no problem...
By the way these are my parameters

```
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["pm.vbat"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 1 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          resetEstimation: 1
      # tracking
      motion_capture_type: "none" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      vicon_host_name: "192.168.10.1:801"
      # optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: True

```

Thanks.

whoenig, 2020-11-08:
The ROS node really just sends the message over the radio to the firmware (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L521-L536). If it works with crayzflie_ros, it must be related to different parameters (e.g., different controller or Kalman gains).

jpreiss, 2020-11-11:
The `velocity_world` command is a velocity in the world coordinate system. You are seeing the expected behavior. It sounds like you want a velocity in the Crazyflie's body coordinate system.

Hurisa, 2020-11-11:
Hi @whoenig I think I should keep the PID controller to send velocity commands as discussed in #209 .
As for the Kalman gains, could you provide some pointers on how to change those?

@jpreiss Is that so? Even if I specify the `frame_id `to be `cf0`? I ask this because that was also my first thought but I have tried sending simple velocity commands in the crazyflie's `x` direction both when it was aligned with the world reference frame and when it wasn't, and it always move forward (in its local frame) ...

whoenig, 2020-11-12:
You can change the gains, for example, in the `crazyflieTypes.yaml`, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L34-L38 for some example settings.

I don't think the `frame_id` is considered in the code - whatever you send is directly packed in CRTP and send to the drone over the radio.

Hurisa, 2020-11-25:
Hi @whoenig 
Sorry for the late reply. I was without access to my lab for a few weeks.
Unfortunately, adjusting the parameters didn't make any discernible difference. Below is what my `crazyflieTypes.yaml` looks like for the `default `configuration.

```
crazyflieTypes:
  default:
    bigQuad: False
    batteryVoltageWarning: 3.8  # V
    batteryVoltateCritical: 3.7 # V
    markerConfiguration: 0 # change to 3 for single marker tracking
    dynamicsConfiguration: 0
    firmwareParams:
      kalman:
        pNAcc_xy: 4.0 # default 0.5
        pNAcc_z: 8.0 # default 1.0
        mNGyro_rollpitch: 0.2 # default 0.1
        mNGyro_yaw: 0.1 # default 0.1
      ctrlMel:
        kp_xy: 0.4 #0.6 # p for position
        kd_xy: 0.2 # 0.35 # d for position
        ki_xy: 0.05
        i_range_xy: 2.0
        kR_xy: 70000 # p for attitude
        kw_xy: 20000 # d for attitude
        kR_z: 60000 # 70000 # p for yaw
        kw_z: 12000 # 30000  # d for yaw
        ki_m_z: 500 #0 #20000     # i for yaw
        i_range_m_z: 1500 # i range for yaw
        kd_omega_rp: 200 # roll and pitch angular velocity d gain
        kp_z: 1.25
        kd_z: 0.4
        ki_z: 0.05
        i_range_z: 0.4
        mass: 0.032
        massThrust: 132000
```
Any ideas are much appreciated.
Cheers

whoenig, 2020-11-26:
I think what you were describing is exactly the expected behavior:

- `velocity_world` is always in world coordinates. The frame-id is ignored.
- The world coordinates are defined by a) boot-up (on-board localization only, i.e., flow-deck), or b) external system (motion capture or similar). Since you seem to operate in mode a), it is expected that the drone moves always in the direction you booted it up in.

If you use the flow-deck, I recommend using the default Kalman gains from Bitcraze. Our (Crazyswarm) adjustments are only useful when operating with a motion capture.

Hurisa, 2020-11-26:
Ok.. so is it possible at all to achieve I am trying to?
Combining linear and angular velocities, by sending the command in the local frame?

Because with `crazyflie_ros,` publishing to `cmd_hover` achieves this already, but that topic doesn't appear when I launch with the crazyswarm. I tired publishing to `/cf0/cmd_vel` but the cf just fell...

whoenig, 2020-11-26:
The proper way would be to simply implement `cmd_hover` in the Crazyswarm. It is easy to do, just wasn't done yet because nobody needed it thus far. `cmd_vel` (despite its name) is very different and should only be used for manual, joystick-based control.

Hurisa, 2020-11-26:
Ok @whoenig, thanks.

Are there any pointers you could give me on how to start implementing the` cmd_hover` in crazyswarm?

Much appreciated

whoenig, 2020-11-26:
- Add a new subscriber, similar to https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L217
- Add a new callback function, similar to https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L521-L536; the function itself will look like the one in crazyflie_ros: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L521-L536
- Everything else should be just fixing up variable declarations, and perhaps adding includes. If you are familiar in C++ this will be easy, otherwise this can be quite some headache of course. 

Let me know if you need help. It would be easy for me, but I think this is actually a great way for the community to step in and to create a pull request (your work is much appreciated!).

Hurisa, 2020-11-26:
Cool! I'm happy to try to contribute! I'll take a look over the weekend trying to see if everything compiles, and early next week do some tests.
I am familiar with C++ but sometimes that doesn't make much of a difference =D

I'll get back to you if I have any major issues, or if I'm able to implement it.

Hurisa, 2020-11-30:
Hi @whoenig 

Just to let you know I was able to make the changes in `crazyswarm` to take in `cmd_hover` type msgs. It compiles and runs as expected!!

However (and this might be a bit off topic), I had forked your repo before and was wondering if it is possible to create pull requests across different forks?
Thanks

whoenig, 2020-11-30:
You always have to fork. Then you push your changes into another branch in your own fork. Then you can create a pull request on github.

whoenig, 2020-11-30:
Ohh and great progress!

whoenig, 2020-11-30:
Looks great - thanks. I'll close this issue now. If there is anything else missing, feel free to re-open.

closed 2020-11-30 by whoenig

Hurisa, 2020-11-30:
Cool. No worries. Glad I could help.