USC-ACTLab/crazyswarm #272: "Exception thrown while processing service call: timeout” error

manikamakam, 2020-11-09:
Hello,

I am trying to use the crazyswarm repo to fly two crazyflies (for now) using VICON system. I have gotten to a place where our ROS system recognizes the existence of crazyflies. I verified this with the provided chooser.py script. 

First issue: If I click reboot, I can observe the crazyflies switching off and turning back on, thus proving that the connection is established. However, if I click on battery, I see 0.00V for both of them though their batteries are charged. I'm not sure why reboot is working but not the battery option. 
[image] 

Second issue: I've selected only one crazyflie from the chooser.py. Before going to VICON, I want to fly the crazyflie with joystick. When I run `roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/2M`, it throws an error saying “Exception thrown while processing service call: timeout”. No values were found in the IMU topic and we can’t see the crazyflie on the RViz display. The uri is correct as I cross-verified by running `rosrun crazyflie_tools scan -v`. I think the connection is established as the reboot is working in chooser.py script. But I'm unsure on why I'm facing this error. 

[image]

Any suggestions to solve these issues would be greatly appreciated. Thank you.

whoenig, 2020-11-09:
1. Did you update your NRF firmware as well as the Crazyradio firmware? Wrong battery measurements usually indicate that there is a mismatch in the versions (the communication protocol changed slightly over the versions).
2. Teleoperation is (currently) not supported with the Crazyswarm. You should use `hover_swarm.launch` instead and navigate using trajectories or waypoints. It would be easy to add Joystick control, but there is just no out-of-the-box example, yet.

manikamakam, 2020-11-13:
Thank you for your response. I've previously updates crazyflies but not the radio. By updating radio, I'm able to get battery values. I have two other issues now. 

1. While I'm building the repository, I get the following 
arm-none-eabi-size cf2_nrf.elf
   text	   data	    bss	    dec	    hex	filename
  63924	    604	   4888	  69416	  10f28	cf2_nrf.elf
S110 Activated
BLE  Disabled
Built for platform cf2
fatal: reference is not a tree: 82e80521124c6e96e5fdb3538a2533f994e2db8f
Unable to checkout '82e80521124c6e96e5fdb3538a2533f994e2db8f' in submodule path 'vendor/FreeRTOS

**I'm able to run all the codes but I was just wondering why is it not able to checkout that specific sub-module. All the other sub-modules are updated.** 

2. I'm running the `roslaunch crazyswarm hover_swarm.launch` command. But I get the below output: 

... logging to /home/ottelab/.ros/log/03bada0c-25da-11eb-8272-9cefd5fcd5b4/roslaunch-ottelab-Precision-Tower-3620-24472.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ottelab-Precision-Tower-3620:43053/

SUMMARY

PARAMETERS
 * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/CF21SingleMarker/bigQuad: False
 * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
 * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
[... 224 lines cut ...]
[ INFO] [1605290387.613378673]: Started 1 threads
crazyswarm_server: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[crazyswarm_server-2] process has died [pid 24512, exit code -6, cmd /home/ottelab/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/ottelab/.ros/log/03bada0c-25da-11eb-8272-9cefd5fcd5b4/crazyswarm_server-2.log].
log file: /home/ottelab/.ros/log/03bada0c-25da-11eb-8272-9cefd5fcd5b4/crazyswarm_server-2*.log

**Actually, one time, I could run without any errors and when I hit the start button in joystick, it did takeoff. But I'm not able to do that again. I end up with above output. I do kill the crazyswarn_server and all other nodes before every run.** 

**My crazyflies.yaml is as follows** 
crazyflies:
- channel: 100
  id: 1
  initialPosition: [0.0, 0.0, 0.0]
  type: default

**allCrazyflies.yaml is as follows:**
crazyflies:
  - id: 1
    channel: 100
    initialPosition: [0.0, 0.0, 0.0]
    type: default

**hower_swarm.launch is as follows:**
[image]

**My crazyflie name in vicon is cf1 and tracking is enabled in vicon. Because I'm using 'motionCapture', I believe I need not set the markers position or anything in this mode. Please let me know why I'm facing the process mentioned earlier. As I was able to fly once and I can see cf1 parameters being loaded, I'm assuming the communication between vicon and my ROS system is established.**

whoenig, 2020-11-13:
1. I don't think you have to worry about that if the compilation works. You can always switch to the latest firmware version, too (`git checkout master`, `git pull` in the crazyflie-firmware folder).

2. I have not seen that before. Which OS version and compiler are you using? From the link in the error message it sounds like an incorrect usage of Eigen somewhere in the code.

jpreiss, 2020-11-15:
I had this submodule error too under Crazyswarm, but I couldn't reproduce it in a standalone clone of bitcraze/crazyflie-firmware. I don't know what is going on.

manikamakam, 2020-11-23:
@whoenig I am using Ubuntu 16.04 with ROS kinetic. I am getting the eigen value error sometimes, but not always. Now, the crazyflie takes off but doesn't hover, instead it's just crashing. What can be the possible issues for this behaviour? 

Also, you mentioned previously, that I can navigate using waypoints. Can you please let me know how to do that? It would help even if you can point me towards an existing tutorial for this. Thanks a lot for the help!

whoenig, 2020-11-23:
Crashes are mostly related to tracking issues. Make sure all firmwares (radio + NRF + STM) were updated. Make sure the coordinate system is correct by moving the drone manually and looking and the `stateEstimate.x/y/z` logging variables.

Perhaps https://drive.google.com/file/d/15favAyrLLpC_O6nrAp-eIbZijFUMLgwV/view helps? There is an example on slide 9 for a simple script. The function you are looking for is called `goTo`.

manikamakam, 2020-11-23:
Thank you that helps! I'm trying to do it first in simulation but I have two errors. 

If I do 
`cd ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim
make
`
I get the following error: 
`swig -python -I../../../../../../crazyflie-firmware/src/modules/interface cffirmware.i
setup.py build_ext --inplace
make: setup.py: Command not found
Makefile:6: recipe for target 'swig' failed
make: *** [swig] Error 127
`
And if I do 
`cd ros_ws/src/crazyswarm/scripts
 python figure8_csv.py --sim
`
I get following error: 
`Traceback (most recent call last):
  File "figure8_csv.py", line 9, in <module>
    swarm = Crazyswarm()
  File "/home/ottelab/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyswarm.py", line 39, in __init__
    from .crazyflieSim import TimeHelper, CrazyflieServer
  File "/home/ottelab/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflieSim.py", line 7, in <module>
    from .cfsim import cffirmware as firm
  File "/home/ottelab/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 28, in <module>
    _cffirmware = swig_import_helper()
  File "/home/ottelab/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 20, in swig_import_helper
    import _cffirmware
ImportError: No module named _cffirmware
`
I refrred to [https://github.com/USC-ACTLab/crazyswarm/issues/47](url) and [https://github.com/USC-ACTLab/crazyswarm/issues/95](url) but nothing worked. There is no crazyswarm-next branch in crazyflie-firmware.

jpreiss, 2020-11-23:
We have now added mixed python 2/3 support so you must specify the environment variable: `CSW_PYTHON=python2 make`

manikamakam, 2020-12-02:
Thanks for he help! 

I observe the crazyflie crashing with both motionCapture and libobjecttracker modes. With motioncapture, after the crazyflie takes off, I get a warning saying that "No pose updated" and with libobjecttracker, it says "[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters."

I have seen in one of the forums that "libobjecttrcaker" is preferred. If so, can you address the error mentioned above. If not, can you give some suggestions on how to fix the motioncapture error?

MauroPfister, 2020-12-11:
> I had this submodule error too under Crazyswarm, but I couldn't reproduce it in a standalone clone of bitcraze/crazyflie-firmware. I don't know what is going on.

**TL;DR**
Update your Git version (using [Git PPA](https://launchpad.net/~git-core/+archive/ubuntu/ppa)) if you are on Ubuntu 16.04.

We could reproduce the same problem on Ubuntu 16.04 but not on Ubuntu 18.04. Turns out that it is an issue related to the Git version. Ubuntu 16.04 ships with Git 2.7 while Ubuntu 18.04 has Git 2.17. As far as I understand the behavior of `git submodule update --depth` slightly changed after Git 2.8 making it possible to fetch a commit that previous to Git 2.8 could not be accessed (see https://stackoverflow.com/a/17692710). After updating the Git version to the newest one available from the [Git PPA](https://launchpad.net/~git-core/+archive/ubuntu/ppa), everything is working as expected.

If this is confirmed by others it might worth adding a note to the documentation or pinning an issue so people find it easily. Also, sorry for being slightly off-topic here.

whoenig, 2020-12-13:
@MauroPfister Thanks for the fix! Yes, adding that to the documentation might help many other people who are using the Crazyswarm on 16.04.

@manikamakam The errors you mention indicate that the tracking didn't work. Usually the key question is if the tracking issues occur before or after the crash. One thing to note is that if you operate right at the limit of your mocap (e.g., markers too small, cameras far away etc.) tracking can be very difficult. The best solution in that case is to use single-marker-mode (with libobjecttracker) and put a single, larger marker on top. Make sure that whatever marker you use, that the marker appears at least as 3 pixels on each relevant motion capture camera. Also can you confirm that you verified the tracking and communication by logging `stateEstimate.x/y/z`?

ZimingR, 2021-08-11:
Dear @whoenig,

I am trying to run a python script to control 4 crazyflies. When I `roslaunch crazyflie_demo a_multi_position1119.launch‘, I encountered the same problem: ’Exception thrown while processing service call: timeout
Service call failed: service [/add_crazyflie] responded with an error: timeout‘
In fact, I can connect to crazyflies through cfclient. When I try to run this script on collegue's system, it succeeded. Any suggestions to solve this issue would be greatly appreciated. Thank you very much!

whoenig, 2021-08-11:
@ZimingR Please open a new discussion topic for your problem. When you do, make sure you include your crazyflies.yaml, as well as the URI (rate, address, channel) of your Crazyflies.

ZimingR, 2021-08-13:
Thanks a lot！

hit-JB, 2021-12-06:
@ZimingR Have you solved the problem that the service throws exception "time out" when  launch the razyflie_demo multi_teleop_xbox360.launch file

ZimingR, 2021-12-06:
Sorry, not yet.