USC-ACTLab/crazyswarm #273: CF2.0 IMU Data

easy98, 2020-11-11:
Hello,

I want to log raw imu data from CF2.0, and how can I direct extract the imu data by crazyswarm?

Thanks in advance.

whoenig, 2020-11-12:
You can enable logging [here](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L48) and add the desired logging variables [here](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L14). The variables you are looking for are `acc.x`, etc. and `gyro.x`, etc., see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L14.

easy98, 2020-11-16:
Thanks Whoenig, it works!

closed 2020-11-16 by whoenig