USC-ACTLab/crazyswarm #277: Simulation in ROS

jpreiss, 2020-11-30:
Crazyswarm's simulation layer is great if your whole project can be expressed as a single `pycrazyswarm` script, but less useful if you want to use Crazyswarm as part of a nontrivial ROS setup. Instead, we would need a ROS node that publishes/subscribes to the same ROS topics and provides the same services as the Crazyswarm server, but instead of actually talking to the mocap and Crazyradio, it does something similar to `pycrazyswarm`'s `CrazyflieSim` logic internally.

This seems like it would be a fairly easy project, any thoughts @whoenig?

whoenig, 2020-11-30:
I think it is a good idea. Some thoughts:

Pros:
 - Works in more standard ROS setups (like you said)
 - Could be extended to interface with simulation for a subset of quadrotors (e.g., Gazebo + RotorS / CrazyS)

Cons:
 - Potentially slower (especially if we do go the Gazebo route, which is slow even if physics is disabled for some quads)
 - Simulation requires ROS (our current solution works without ROS for testing high-level scripts)

jpreiss, 2020-12-01:
I should have clarified, I am not proposing this as a replacement for `--sim`. I was thinking more like a [launch file command line arg](http://wiki.ros.org/roslaunch/XML/arg) for the crazyswarm server that causes it to run the simulation logic instead of connecting to the hardware.

jpreiss, 2020-12-01:
If we port `CrazyflieSim` to C++ we wouldn't need any physics simulator. It only simulates an integrator. However, we would need to compile parts of the Crazyflie firwmare into the Crazyswarm server for things like piecewise polynomials and the onboard collision avoidance.

closed 2021-02-20 by jpreiss