USC-ACTLab/crazyswarm #280: There is no noetic version for vrpn

xyyjcyd, 2020-12-07:
Now I am trying to run crazyswarm based on ubuntu20.0.4 & python3.8 & ros-noetic. When I tried to install the vrpn for distro ros-noetic, according to ROS index, I found there is still no version for Noetic. 
[https://index.ros.org/r/vrpn/#noetic](url)
[image]

jpreiss, 2020-12-07:
See discussion in https://github.com/USC-ACTLab/crazyswarm/issues/248. It appears that the ROS vrpn package doesn't support Noetic yet.

Crazyswarm interacts with the mocap system via https://github/com/USC-ACTLab/libmotioncapture. I believe libmotioncapture only needs the VRPN library, not the ROS package. In the past we used the ROS package as a  convenient way to get the VRPN library installed. You should be able to use VRPN by installing https://github.com/vrpn/vrpn such that CMake can find it. @whoenig is this correct?

whoenig, 2020-12-07:
That's accurate - thanks!

xyyjcyd, 2020-12-08:
> See discussion in #248. It appears that the ROS vrpn package doesn't support Noetic yet.
>
> Crazyswarm interacts with the mocap system via https://github/com/USC-ACTLab/libmotioncapture. I believe [...]

Thanks for your help!!

lkyfly0765, 2020-12-08:
Actually, the natnet is more stable and easy to configure. See this [https://github.com/whoenig/NatNetSDKCrossplatform](https://github.com/whoenig/NatNetSDKCrossplatform)

jpreiss, 2020-12-10:
@xyyjcyd were you able to get everything installed OK? If so, please close this issue.

closed 2020-12-13 by xyyjcyd