USC-ACTLab/crazyswarm #281: Enable github discussions

jpreiss, 2020-12-10:
Github added a new feature [Discussions](https://docs.github.com/en/free-pro-team@latest/discussions/quickstart)

> for conversations that need to be transparent and accessible but do not need to be tracked on a project board and are not related to code.

This sounds ideal for "help, it doesn't fly!" issues that are not likely to result in any code changes. We could direct this kind of question to discussions. Issues could be reserved for 1) true bugs that are probably not configuration errors, and 2) new feature proposals.

@whoenig do you agree?

jpreiss, 2021-02-02:
@whoenig I think a lot of the recent flurry of issues would be more appropriate for discussions...

whoenig, 2021-02-03:
Good idea - I just enabled it. I never tried it before, but lets see how it works out.

closed 2021-02-03 by whoenig