USC-ACTLab/crazyswarm #283: About logging LPS raw data

easy98, 2020-12-19:
Hi,

I am currently setting up the LPS system based on #108 using the TWR mode. My purpose is to get the LPS raw data, which seems to be the distance between the tag and anchors [here](https://github.com/bitcraze/crazyflie-firmware/blob/13aa1372bc3ba5b14922fc2f9d8999270882021a/src/deck/drivers/src/lpsTwrTag.c#L569-#L593). I would like to directly log it in [this line](https://github.com/USC-ACTLab/crazyswarm/blob/ab0fe73a05b6e947d5651c411634eb49a3862efc/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L13).  
But when I run hover_swarm.launch, I always get "runtime error". I am not sure if it's the right way to log that data and if it's possible to to that?

Thanks in advance.

whoenig, 2020-12-21:
If you want to log the data in real-time over the radio, this is the correct approach. All you need to do is figuring out the logging variable of interest - in your case `ranging.distance0` etc. If you just want to look at the data in post-processing, I recommend using the uSD-card deck instead (you will be able to log more variables at a higher rate compared to the over-the-radio mode).

If you use the LPS, the Kalman filter is always active automatically, so you should be getting the state as predicted by the on-board filter.

easy98, 2020-12-22:
It works! Thanks Whoenig！

easy98, 2021-01-01:
Hi, 

When I switched the mode to TDOA3 on the python client, I changed the logging variable to "tdoa3.tdoa". But when I run "roslaunch crazyswarm hover_swarm.launch", It throws an error saying “Could not find tdoa3.tdoa in log toc!”. Then I rechanged the logging variable to "ranging.distance0", no error occurred but the value of "ranging.distance0" was always zero and the mode was actually TDOA3 rather than TWR. I am not sure why I am facing this error.

closed 2021-02-20 by jpreiss