USC-ACTLab/crazyswarm #285: CrazyFlie Bolt compatability

annesteenbeek, 2021-01-14:
Hello,

I'm starting research on a new project, combining drone swarms and dance.
There are already UAV frames available, though some problems with the controllers.
I was hoping to replace the FCU with the [CrazyFlie Bolt](https://www.bitcraze.io/products/crazyflie-bolt/).
However, I can't find any information in the documentation on the compatibility with this board.
I know the Bolt is still in early access, but is this something that would be supported or would it require additional work from my side to flash the swarm firmware on this board?

whoenig, 2021-01-14:
It is supported, no custom firmware needed. Let us know if you run into any issues!

annesteenbeek, 2021-01-14:
Great, good to know.
Thanks for the fast response.

closed 2021-01-14 by annesteenbeek