USC-ACTLab/crazyswarm #286: Do I need to use the Natnet and vrpn when using optitrack?

LiuZesensengsheng, 2021-01-19:
Hello!
I want to use the Optitrack as the mocap of crazyflies. but I get really comfused that the instructions said that i only need to modify the hover_swarm.launch while others always mentioned I need to use the vrpn and natnet to keep a connection. What's more, I didn't see the in the code of .cpp
So What should I do to connect the crazyflie to the Optitrack?
Do I need to install both the vrpn and natnetSDK or just modify the hover_swarm.launch?
Thanks

whoenig, 2021-01-19:
The Crazyswarm already comes with the NatNetSDK, so you only need to adjust the launch file. The code is 'hidden' in a submodule, see https://github.com/USC-ACTLab/libmotioncapture/.

LiuZesensengsheng, 2021-01-22:
> The Crazyswarm already comes with the NatNetSDK, so you only need to adjust the launch file. The code is 'hidden' in a submodule, see https://github.com/USC-ACTLab/libmotioncapture/.

Thank you for your advise! Now I get get the message given by the motive and see them in the rostopic
But there comes new question that when I test the script named nicehover. The crazyflie flipped at once. After I choose the  estimator 1 in hover_swarm 21line . The crazyfile can take off but fly randomly. Finally, it always landed and flipped in the ground.  I guess there are some errror in the coordinate with the crazyflie and the optitrack. But after test, it doesn't work. 
Buy the way, I use just one mark in the point cloud.
Thanks one more.

whoenig, 2021-01-22:
This is usually an indication of the coordinate system not matching. Did you follow the settings here: https://crazyswarm.readthedocs.io/en/latest/usage.html#optitrack? Also make sure that when you just manually walk around with the drone, that the state estimate (`stateEstimate.{x,y,z}` logging variables) matches the ROS coordinate system.

LiuZesensengsheng, 2021-01-24:
Thank you for your advise ! 
But it seems doesn;t work. I are using the Ubuntu 20.04 and the python 3 and I get trouble to update the hardware of crazyradio into 99.55 that when the version is 99.55, I can't find the crazyflie. So I have to use the version in 18.0.
 And while I set the enablelogging:true, crazyflie doesn't work so I can't get the estimate msg. After I read the code, I find that the node in the ground didn't set the messige of Position to the crazyflies. The software get the data from the optitrack and set it to the pointcloud, the pointcloud set the msgpointcloud to the RVIZ and RVIZ calculate the msg and publish it to 
 /tf.
I'm using the singlemarker, so the mative can;t give the pose of crazyflie of cousely. So how can the crazyflie eminate its yaw? I can't find the EKF algorithm in the code. And now my crazyflies can take off but can't hover in a place. It just fly toward a random direction and go straight line to the wall.
sincere thanks!

whoenig, 2021-01-24:
**Crazyradio** I believe the update procedure requires Python 2, which is not shipped by default in Ubuntu 20.04. If you bought your Crazyradio recently, you likely do not need to update your radio firmware. Otherwise, please open a new issues with the details on how the radio firmware upgrade failed.

**Logging** Could you provide some more details on how enabling logging did not work? The logging is crucial for debugging your issue, to avoid (or at least minimize) the crashes

**Tracking** On the PC-side the pointcloud is analyzed every frame with an optimal assignment algorithm to track the position of each Crazyflie. This is similar to GPS on a bigger drone flying outdoors. The EKF runs onboard the firmware and fuses additional sensor information from the IMU to estimate the orientation. You may find the code at https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core.c.

LiuZesensengsheng, 2021-01-27:
Thank you for you advice!
I turn the firmware version back into the 0.53 of crazyradio and the crazyflie is 20209. And i think it does not matter.
And In my many times test, I found that the position from the optitrack seems take no effert to the crazyflies. My hover_swam.launch is as follow:
`<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator:  2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16    # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          resetEstimation: 1
      # tracking
      motion_capture_type: "optitrack"         # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      send_position_only: False                # set to False to send position+orientation; set to True to send position only
      optitrack_host_name: "192.168.50.223"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  
  

</launch>`
The land function also doesn't take effert. 
Now I'm using a flowdeck sensor in the Crazyflies and it can run the test script. Goto z=1 and goto z=0.4. or goto x=1 y=1 and reture x=0,y=0, it works well. I use the ekf to estimater the position. But data flow from the optitrack seems  useless, I try to use the fake date such as put a label in the grown and let the crazyflie run with the  wrong data. However, it still work with the help of flowdeck. And when I turn off the flowdeck, the crazyflie go randomly again. So I get comfuse that my optitrack didn't work in the EKF algorithm.
Thank for your help(and it seems that  I need more （*^_^*))

LiuZesensengsheng, 2021-01-27:
> **Crazyradio** I believe the update procedure requires Python 2, which is not shipped by default in Ubuntu 20.04. If you bought your Crazyradio recently, you likely do not need to update your radio [...]

Now I suspect that the place position from optitrack was not sended to the crazyflie. And the EKF algorithm may has some error ?

whoenig, 2021-01-27:
From your description it sounds like you can see the state estimate in ROS topic/rviz, but it does not get transmitted correctly to the Crazyflie. The most common reason for that is i) you didn't update the Crazyradio firmware, or ii) you didn't update the NRF51 firmware. Both firmwares need to support broadcast communication. The best way to test this is by logging the `stateEstimate.{x,y,z}` variables and moving the CF around without flying.

closed 2021-02-20 by jpreiss