USC-ACTLab/crazyswarm #288: How to set nano-seconds as stamp parameter in go_to services?

akicludy2017, 2021-01-29:
Hi, 
I'm using crazyswarm with optitrack system. And I control crazyfiles via ros services like takeoff, go_to, land.
When I call go_to service with only 'nsecs' in 'duration', cf flips down.
ex) duration: secs: 0 nsecs: 500000000
It works fine when I call them with 'secs'.
ex) duration: secs: 1 nsecs: 0
How should I set float seconds, like 0.5 sec, as duration?

jpreiss, 2021-01-29:
`CrazyflieROS::goTo()` uses `ros::Duration::toSec()` to convert the message to a floating-point value in seconds:

https://github.com/USC-ACTLab/crazyswarm/blob/1620165dd69894c5176e1679502b54561091df15/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L446

`ros::Duration::toSec()` does a proper conversion to floating-point ([source](http://docs.ros.org/en/latest/api/rostime/html/duration_8h_source.html#l00093)):

       double toSec() const { return static_cast<double>(sec) + 1e-9*static_cast<double>(nsec); };

the CRTP radio protocol uses a floating-point duration value:

https://github.com/bitcraze/crazyflie-firmware/blob/d91290fd19cff49c75bc442cad112b32b5b91926/src/modules/src/crtp_commander_high_level.c#L211

as do all the further onboard processing steps, so there are no more conversions.

0.5 seconds is a pretty fast duration. Unless you are moving a very short distance, it might require faster acceleration than the Crazyflie is capable of producing.

akicludy2017, 2021-01-30:
Hi James,

Thanks for your quick response.

I've checked duration value during the calls. It looks fine, converted into correct float value.
And I've done more tests with other sets of parameters... It looks like it depends a pair of goal/duration. 
It flies with duration:0.7 seconds as expected. But cannot with duration:0.5.
I think that you're right. 0.5 seconds is too short for go_to command to start moving, stop at the goal, stay hovering. 

And it may also depend on the payload. In my case, cf carries more payload, markers for mocap.  With lighter payload, it can flies with duration:0.5 as expected.

So now. If I wanna control cf in higher frequency, I shouldn't use go_to, which tries hover-move-hover in short period. I should use cmd_vel directly, or upload/start trajectory...?

Anyway, thanks again for your great input. :)

jpreiss, 2021-01-30:
Like you said, `goTo` plans ahead to stop at the goal. If you want to move through a sequence of points without stopping, `goTo` is not a good tool.

Both of the alternatives you mentioned could work. What is the high-level goal? Can you describe how you want the crazyflies to move over a longer time period?

akicludy2017, 2021-01-30:
We're working on kind of swarm controller. We control each vehicle based on others' states, pose, velocity, acceleration, etc.
Each time we got new state from a vehicle, we calculate where each vehicle stay to keep them as one swarm, then let it move.
Currently we're moving them with the same frequency which we get new positions from the lps/mocap system.

By the way, I'm trying cmd_position, which may be easier than cmd_vel. :)

jpreiss, 2021-01-30:
Yes, I would try `cmd_position`.

You might also switch your controller from `mellinger` to `pid` in your `hover_swarm.launch` file. The `mellinger` controller is tuned for smooth pre-planned trajectories, and may have trouble with inputs that jump around a lot. `pid` is lower gain.

akicludy2017, 2021-01-30:
Thanks for your advice. 
I'll try it. :)

closed 2021-02-20 by jpreiss