USC-ACTLab/crazyswarm #289: The motor has inconsistent performance

swarm5, 2021-02-01:
I found that the crazyflie I purchased has inconsistent motor performance. How can I adjust it?
a. When the motor performance is relatively close, cf can realize hovering stably. The pwm of the four motors is shown in the figure.
[image]
b. When the motor performance gap is very large, I use the niceHover program, and you can see a large gap between the four pwm curves in the log. At the same time, as the battery power decreases, cf begins to spin in the air.
[image]
I have a few questions to consult:
1. Can I adjust the PID and other parameters to achieve stable cf flight when the motors are inconsistent?
2. I have plans to customize a batch of 716 motors. What is the rated speed of this motor?
3. I recorded m1-m4 data and pwm1-4 data in the log. I am not very clear about the meaning of m1-4 data? Judging from the data curve, it is similar to pwm, but not exactly the same.

swarm5, 2021-02-01:
@jpreiss @whoenig I want to talk to you, I seem to have found the reason why the plane will be unstable in the air: the inconsistency of the motor leads to the deviation of the heading angle of cf.
1. In CF21SingleMarker mode, all aircraft that fly unstable will have the problem of heading angle drift.
2. In another experiment, I used the same aircraft. In the default mode, I built the marked point as a rigid body and gave cf heading feedback so that cf could fly stably.

I have a few questions to ask you:
1. (Four-point rigid body type) If I use the default mode, all the marked points are established as rigid bodies. How to set dozens of different rigid body styles? What experience does everyone have?
2. (Four-point tracking type) If I install markers on the same four positions of cf and set it to four-point tracking mode, will the system give feedback to cf heading angle information in this case?
3. (Single point tracking type) Is there any other way to solve the course deviation problem in single point tracking mode? It seems that calibrating the gyroscope is a way to try. How does cf calibrate the gyroscope (connect to cfclient, will there be instructions?), how often do you need to calibrate the gyroscope? Do you have experience?

jpreiss, 2021-02-02:
1. If you arrange the markers differently on each aircraft, then you need to create the objects manually in your motion capture software. The Crazyswarm server cannot programmatically create objects.
2. I am not sure what you are asking here. How is the setup in this question different from your experiment 2?
3. Integrating the gyroscope signal is not a viable way to estimate heading. It is not accurate enough and too noisy. In single-marker mode, you must move the aircraft around to make the heading observable. For example, you could "orbit" in a small circle around the hover point.

In theory, the onboard Kalman filter could use the IMU compass to correct the gyroscope drift, but it does not. I am not sure why.

swarm5, 2021-02-02:
Four-point tracking type, I mean that all cfs are equipped with four mark points, but the positions of all mark points are the same, they cannot form rigid bodies with different styles. In this case, can cf get heading feedback?

swarm5, 2021-02-02:
I want to know how to calibrate cf's gyroscope and accelerometer? Is it auto-calibrate after booting? Is it necessary to connect to cfclient? Does it require other special operations?

whoenig, 2021-02-02:
* Four-point tracking: yes, heading (and roll/pitch) are computed.
* IMU compass: There is none in the CF 2.1. For CF 2.0, the measurements were apparently affected significantly by the motors spinning. The gyro in CF 2.1 drifts less and you should be able to hover at a point for about 30 seconds or so.
* Calibration: The gyro bias is calibrated on boot-up automatically - no special setup is needed (just make sure you don't move the CF around). The accelerometer bias is not measured by default, because it would require a perfectly flat ground at start-up. I believe there is a firmware flag to enable this calibration though.

swarm5, 2021-02-03:
Our laboratory uses cf2.1, only one marker is installed, and the cf type is CF21SingleMarker. **The problem we encountered is: before cf takes off, if the orientation of cf can perfectly coincide with the x-axis, cf can fly better. However, if the cf heading angle will be offset, the flight of cf will be unstable.**
During the test, we also found a phenomenon: **cf heading shifts, this situation mostly occurs during cf takeoff or landing. We can improve this problem by reducing the maximum flight speed in the z direction.**

I have a few questions to ask:
1. Have you encountered any heading deviations that occur during takeoff and landing? Do you have other experience in handling?
2. I am not quite clear about the principle of single-point state estimation. Why can the drift of the gyro be corrected when the x-y plane moves horizontally? Have you published relevant papers to introduce a single landmark for pose estimation?
3. Or have you published a paper introducing the control algorithm of cf in crazyswarm? Because I want to try, it is more stable to fly with a single mark by adjusting the parameters. For many parameters in crazyflieTypes.yaml, I am not sure about their specific meaning, for example:
 ```
i_range_m_z: 1500 # i range for yaw, kalman:
        pNAcc_xy: 1.0 # default 0.5
        pNAcc_z: 2.0 # default 1.0
        mNGyro_rollpitch: 0.1 # default 0.1
        mNGyro_yaw: 0.1 # default 0.1
```

whoenig, 2021-02-03:
1. I have not seen that, although most recently I have only flown with a non-standard controller.
2. This is just a GPS-like update of the Extended Kalman Filter (EKF). A good intro is here: http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf. The actual EKF is implemented here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core.c, with academic paper references at the top of the file.
3. The variables you are interested in are described in the papers of the above file. The controller is at https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/controller_mellinger.c, with academic references at the top of the file.

closed 2021-02-07 by swarm5