USC-ACTLab/crazyswarm #291: How to determine the yaw angle of cf

ucaszzw, 2021-02-02:
I encountered a very difficult problem. 1. When I use the CF21SingleMarker type, some cfs will have a heading shift, and then spin in the air. 2. When I use the rigid body type, I cannot realize the arrangement of 20-30 different rigid bodies in such a small volume as cf. Does anyone have any good suggestions?
The CF21SingleMarker type is indeed very convenient and robust, but if there are deviations in the motor, propeller, etc., it is easy to cause the heading of cf to shift, and then cause cf to spin in the air or fly unstable.

ucaszzw, 2021-02-02:
Is there a simpler way (not to build so many rigid bodies) to obtain the heading angle information of cf?

whoenig, 2021-02-02:
If you use 4 markers and `libobjecttracker` tracking type, then all CFs can share the same marker configuration. Tracking is done by our library using ICP, and not by your motion capture system software (which will just provide a pointcloud of all markers).

If you use the single marker mode, make sure that you place the Crazyflie initially with the correct yaw. Also, flying in the x/y plane will help the EKF to recover the yaw angle (i.e., avoid hovering for a long period of time in single marker mode). The Crazyflie 2.1 has an improved IMU (over 2.0), so the yaw-drift over time is significantly lower.

closed 2021-02-02 by whoenig

ucaszzw, 2021-02-02:
Thank you very much for your answer @whoenig . Can I understand it like this, if I use 4 markers and libobjecttracker tracking type,
1. Compared with rigid bodies, it does not require so many styles.
2. Compared with a single point, cf will receive accurate heading information?
I want to know, 4 markers and libobjecttracker tracking type, are there any disadvantages of this type?

whoenig, 2021-02-02:
The disadvantage is that it a) requires an initial guess of the position of all Crazyflies, b) does not handle occlusions well. Both stem from the fact that we do frame-by-frame tracking. In practice, this also means that you do need a high-quality motion capture system. If your motion capture has lots of spurious markers or frequently does not see all the markers, the system will be less robust compared to the single-marker tracking mode.

ucaszzw, 2021-02-02:
@whoenig Thank you for your introduction. It seems that single-point tracking mode is the best choice. However, we have done a lot of tests. In the case of using the CF21SingleMarker type, it needs better motor consistency, otherwise it will fly figure8 over and over while spinning in the air.
Do you think it is necessary to adjust the kalman parameter or the ctrlMel parameter in firmwareParams?

whoenig, 2021-02-02:
Both are already tuned, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L64-L86. However, if you use a Crazyflie 2.0, you should use `defaultSingleMarker` instead, which uses different tuning parameters (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L34-L56).