USC-ACTLab/crazyswarm #334: Problem connecting to optitrack mocap

manvibharat, 2021-02-26:
I have tried https://github.com/whoenig/NatNetSDKCrossplatform and ./mcr works well and I am getting the data there. But when I run mocap_helper.launch with the following contents

<?xml version="1.0"?>
<launch>

  <node pkg="crazyswarm" type="mocap_helper" name="mocap_helper" output="screen" >
    <rosparam>
      # tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_host_name: "optitrack"
      #optitrack_local_ip: "192.168.43.238"
      #optitrack_server_ip: "192.168.43.73"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
    </rosparam>
  </node>
</launch>

 it gives the following error

started roslaunch server http://rosmelodic:40259/

SUMMARY
========

PARAMETERS
 * /mocap_helper/motion_capture_type: optitrack
 * /mocap_helper/object_tracking_type: motionCapture
 * /mocap_helper/optitrack_host_name: optitrack
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

ROS_MASTER_URI=http://localhost:11311

process[mocap_helper-1]: started with pid [23540]
frame 0:
[mocap_helper-1] process has died [pid 23540, exit code -11, cmd /home/rosmelodic/crazyswarm/ros_ws/devel/lib/crazyswarm/mocap_helper __name:=mocap_helper __log:=/home/rosmelodic/.ros/log/03efa038-7851-11eb-88d8-c4041567d7d3/mocap_helper-1.log].
log file: /home/rosmelodic/.ros/log/03efa038-7851-11eb-88d8-c4041567d7d3/mocap_helper-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

However when I use methods described in some of the issues like using local ip and server ip of optitrack nothing works terminal stays as it is. So what can be the solution to this.

The ./mcr output is as follows

Begin Packet
-------
Message ID : 7
Byte count : 96
Frame # : 11627530
Marker Set Count : 0
Unidentified Marker Count : 0
Rigid Body Count : 1
ID : 1
pos: [1.60,-1.17,-0.00]
ori: [-0.00,-0.00,-0.17,-0.98]
software latency : 0.000
Timestamp : 0.000
Mid-exposure timestamp : 4530722105326567424
Camera data received timestamp : 0
Transmit timestamp : 4682726790983896269
End Packet

Also I downloaded NatNetSDK from official source and placed in external dependencies but no luck with that.

So any solutions possible?

whoenig, 2021-02-26:
How did you run `./mcr`, since that requires an IP address to be provided? Local and server IP are not used anymore. The only parameter you need to set is optitrack_host_name, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/mocap_helper.cpp#L51.

whoenig, 2021-02-26:
You don't need to download NatNetSDK manually. Make sure you use the version we tested with, that is included as a git submodule. The crash you are observing is likely related to some change in the packet format.

manvibharat, 2021-02-26:
> How did you run `./mcr`, since that requires an IP address to be provided? Local and server IP are not used anymore. The only parameter you need to set is optitrack_host_name, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/mocap_helper.cpp#L51.

Yeah I used the ip address of the host which I saved in /etc/hosts

whoenig, 2021-02-26:
You should use the same argument in mocap_helper; just because it is called hostname, doesn't mean that it can't take an IP address.

manvibharat, 2021-02-26:
@whoenig 
Also if it helps I am using Optitrack 1.10.3 version. Do you think it is the problem?. Also if possible can you suggest which file I should look to change for this to change packet format. 
Thanks

whoenig, 2021-02-26:
We only tested with Motive 2.X. For the packet format, it would be best to find out where the crash occurs (see http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB on how to run a ROS node with a debugger). Then you can compare the packet decoding in that part of the code with the one that is used in the NatNetSDK example.

closed 2021-02-27 by jpreiss