USC-ACTLab/crazyswarm #342: Install dependencies locally with github actions

jpreiss, 2021-03-05:
The github actions CI scripts are a better source of truth for system setup than hand-written docs because they are constantly tested.

- [ ] Figure out if there is an easy way to execute a github action locally.
- [ ] If so, add instructions to docs.
- [ ] Deprecate manual setup instructions in docs OR add warning about being out of sync.

ataffanel, 2021-03-08:
This is a good idea but the only way I have found to run github action locally is [act](https://github.com/nektos/act) and it runs the action in a docker image to emulate github's hosted runner. This makes a lot of sense to run builds but does not help much to install dependencies.

What I have seen done in other project it to put the commands in a shell script and run the script from CI, it is not as clear as listing the commands in the CI configuration but it still has the advantage of having a single source of truth tested on each commits.

jpreiss, 2021-05-05:
@ataffanel I would be OK with loss of clarity, but currently we rely on the `ros-tooling/setup-ros@v0.1` action to install ROS, which would be a bit of a pain to reproduce in a shell script.

jpreiss, 2021-09-29:
not currently planning to work on this.

closed 2021-09-29 by jpreiss