USC-ACTLab/crazyswarm #343: Unable to checkout '82e80521124c6e96e5fdb3538a2533f994e2db8f' in submodule path 'vendor/FreeRTOS'

WaterHyacinthInNANHU, 2021-03-05:
Hi!
When I excute `./build.sh` it kept throwing the following exception:
```
fatal: reference is not a tree: 82e80521124c6e96e5fdb3538a2533f994e2db8f
Unable to checkout '82e80521124c6e96e5fdb3538a2533f994e2db8f' in submodule path 'vendor/FreeRTOS'
```
Maybe there is a new unexpected commitment on FreeRTOS repo?
Please, could anyone help me out?

jpreiss, 2021-03-05:
try checking out the last commit from February instead of the most recent one. We just updated the crazyflie firmware version and there is something weird about the way it uses the FreeRTOS submodule.

jpreiss, 2021-03-05:
Appears to be an instance of https://github.com/bitcraze/crazyflie-firmware/issues/620

WaterHyacinthInNANHU, 2021-03-06:
> try checking out the last commit of crazyswarm from February instead of the most recent one. We just updated the crazyflie firmware version and there is something weird about the way it uses the FreeRTOS submodule.

Sorry for getting back to you late. Thanks for your timely reply and useful suggestions.
I'm runing Ubuntu16.04 with git version 2.7.4.
I've tried checking out to `65413e53ea0a6ed9c38a4d45ff4382b76d861fef` and some solutions in [bitcraze/crazyflie-firmware#620](https://github.com/bitcraze/crazyflie-firmware/issues/620), but neither solved it.  
So I cloned and checkout [this commit of FreeRTOS](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/3604527e3b31991c596cd420f32989ee890aca4a) under `vendor/FreeRTOS` and break `./build.sh` down to run it step by step. It fixed my problem.
I'll run further tests while using it.
Thanks again.

whoenig, 2021-03-06:
I believe this was [recently fixed](https://github.com/bitcraze/crazyflie-firmware/commit/37b9b1712b75a968cf362ab4341afdfa9e9c1bd0) in the official firmware. Note that you will need `git submodule sync` if you update an existing repository.

WaterHyacinthInNANHU, 2021-03-06:
> I believe this was [recently fixed](https://github.com/bitcraze/crazyflie-firmware/commit/37b9b1712b75a968cf362ab4341afdfa9e9c1bd0) in the official firmware. Note that you will need `git submodule sync` if you update an existing repository.

Thanks! I tried to clone and build again and it works just fine!

closed 2021-03-06 by WaterHyacinthInNANHU