USC-ACTLab/crazyswarm #344: Could not find stateEstimate.z in log toc!

WaterHyacinthInNANHU, 2021-03-08:
Hi!
When I ran `hover_swarm.launch` I got the following error message:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find stateEstimate.z in log toc!
```
The only things I modified are setting `enable_logging` to `True`and `genericLogTopic_log1_Variables` as ` ["stateEstimate.x", "stateEstimate.y", "stateEstimate.z"]` in `hober_swarm.launch`

1)Actually, whenever I set `stateEstimate.z` in `genericLogTopic_log1_Variables` I got this error. Things were all good without logging it. I believe there's something wrong with my configurations cuz when I ran crazyswarm on another laptop it never happens.

2)BTW, how exactly to view the "current value" as mentioned [here](https://github.com/USC-ACTLab/crazyswarm/issues/170#issuecomment-536661794)? I tried `rosparam list` but no entry in output seems to be related with 'stateEstimate'. I'm new to ROS-_- , maybe there's a better way to check it?

whoenig, 2021-03-08:
First check if you have stateEstimate.z in your firmware. You can run `rosrun crazyflie_tools listLogVariables --uri <URI>`.

For the current value, you'd use `rostopic echo /cf<id>/log1`. This topic just contains a generic array - the values correspond to the logtopics as specified.

WaterHyacinthInNANHU, 2021-03-09:
Thanks for your reply! I'll double check my firmware later.

sabdi1, 2021-03-12:
Hello,

I am having this issue with logging as well.
I've checked my firmware via the proposed rosrun crazyflie_tools listLogVariables and I do have stateEstimate.x,y,z.
However, when I attempt to add them too genericLogTopic_log1_Variables: [], I also get this error when launching hover_swarm.launch.

Do you have any more advice on how to remedy this Whoenig?

whoenig, 2021-03-12:
@sabdi1 Could you specify which exact firmware you are running? I think I'll need to try to reproduce this issue - I have not seen that problem before.

sabdi1, 2021-03-12:
I have the latest firmware: 2021.01

manikamakam, 2021-03-24:
Hello @whoenig, I'm Manika. Me and @sabdi1 work together. We are still facing the problem with stateEstimate.z. I have set both enable_logging_pose and enable_logging to True and I get an error saying

what(): Could not find stateEstimate.x in log toc!

I'm wondering if there is any update regarding this issue. Thank you.

whoenig, 2021-03-24:
I just tried the same setup: 1 CF with firmware 2021.01, motion_capture set to None, logging enabled, logging pose enabled. I cannot reproduce the issue you are seeing.

Is your repo up-to-date, i.e., did you do a fresh checkout or use

```
git submodule sync
git submodule update --recursive
```
?

sabdi1, 2021-04-02:
I've tried doing 
`git submodule sync
git submodule update --recursive`
I got the same error. 

Then I cloned the repository again and did a fresh build. I have the same error. When I run `rosrun crazyflie_tools listLogVariables --uri <URI>`, I have stateEstimate.x and stateEstimate.z in the list. However, if I run `rostopic echo /cf<id>/log1`, I don't get anything. 

Please let me know what next steps I can take to get past this error.

sabdi1, 2021-04-02:
Attached is my hover_swarm.launch file for reference.

[hover_swarm.zip](https://github.com/USC-ACTLab/crazyswarm/files/6250808/hover_swarm.zip)

whoenig, 2021-04-05:
I just tried the same: 1x Crazyflie 2.1 with 2021.03. Crazyswarm freshly checked out and rebuilt in a new folder using your launch file. I do not get any errors. If I run `rostopic echo /cf<id>/log1` I get an empty list (since you didn't configure any logging variables). If I use `rostopic echo /cf<id>/pose` I get the full pose estimate as computed by the EKF onboard. The only difference I see is that I don't have an LPS deck (I use a flow deck instead to mimic the "none" type). I am using Ubuntu 20.04. Just to be sure, you are still getting the same `std::runtime_error` error? Could you post the full output log?

sabdi1, 2021-04-09:
Hello @whoenig,

Thank you for your inputs. Here are some observations:
1. I have tried using only flow deck and removed the LPS deck. But I get the same error.
2. I am using Ubuntu 18.04 with ROS Melodic and python 2.7.17.
3. I have attached a screenshot of the error and also corresponding log files. 
[image]

[79e51b58-9942-11eb-bba3-40ec99c2179e.zip](https://github.com/USC-ACTLab/crazyswarm/files/6286594/79e51b58-9942-11eb-bba3-40ec99c2179e.zip)

Looking forward to your inputs. Thank you.

whoenig, 2021-04-15:
You could try setting https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L46 to True, or delete the cache files in the `~/.ros` folder, as it looks like its loading the names from the potentially incorrect cache. If that doesn't help, you would need to debug the content of the log toc.

sabdi1, 2021-04-23:
Hello @whoenig,

Thank for your help. I am able retrieve the pose information now. I didn't do any changes. I just updated the crazyflie to 2021.03 again, but I'm pretty sure it was the same version earlier too. Anyway, thank you!

whoenig, 2021-04-25:
Strange - but thanks for letting me/us know!

closed 2021-04-25 by whoenig