USC-ACTLab/crazyswarm #348: No Updated Pose for Motion Capture Object

elim17, 2021-03-21:
Hello,

Our `mocap_helper.launch` is working and streaming motion-capture data via NatNet.  We can see the correct changes in pose. However, when running hover_swarm.launch we receive these warnings: 
[image]

The "No Updated Pose for Motion Capture Object" warning is most troublesome to us because the drones will take off but flight is unpredictable and ends with them running into a wall. Any suggestions on where this warning is coming from? It looks like motion capture was not implemented in previous issues where this error was received(2018-2019 issues). From other conversations it looks like motion capture was implemented since then with motive Optitrack.

whoenig, 2021-03-21:
Could you provide a screenshot/printout of `mocap_helper.launch`? My initial guess is that there is a mismatch in the naming of the rigid body in motive (needs to be `cf2` for `hover_swarm.launch` to work, while `mocap_helper.launch` simply prints all data it receives).

elim17, 2021-03-23:
Unfortunately, It seems like my naming convention is correct. This is the output for `mocap_helper.launch`. 
[image]

I did make the cf rigid bodies in Motive, in order as it says to do on your GitHub. This is the test output with both cf1 and cf2 selected. 
[image]

Then I tried `hover_swarm.launch` with both cf1 and cf2. I get an output warning for both. 
[image]

What would be your next guess?

whoenig, 2021-03-23:
Is your tracking type set to motionCapture (https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L36)? If yes, please share both your mocap_helper and your hover_swarm launch files.

elim17, 2021-03-24:
Yes, our tracking type is set to motionCapture.

Below are the mocap_helper and hover_swarm launch files. 
[image]

[image]

whoenig, 2021-03-24:
This is very odd, but also probably a code path that wasn't tested in a long while (since using rigid bodies is not that useful for larger teams). One thing that would be helpful is too add `ROS_INFO("rb %s", rigidBody.name());` in https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1123. I reviewed the code and didn't see any obvious mistake, but this would at least tell us if the rigid bodies are correctly identified in at least some of the frames. Don't forget to use `catkin_make` after changing the code.

elim17, 2021-03-25:
You were right! The rigid bodies don't seem to be correctly identified as were are receiving what looks like empty string. 
[image]
[image]

jpreiss, 2021-03-25:
The `?????` output is probably a null pointer, not an empty string.

elim17, 2021-03-25:
Ok, is that what you were expecting to get from the previous post? It seems like either way the rigid body is not being identified.

elim17, 2021-04-05:
@whoenig Do you have any suggestions based on the null pointer issue?

whoenig, 2021-04-05:
Not sure. You could try adding some more debugging prints here https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1707, to see if directly after the call to libmotioncapture the data itself is valid or not. It looks like some sort of memory corruption, but I am not sure on which level it occurs. Another option is to run the crazyswarm server with valgrind (http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB).

elim17, 2021-04-12:
After debugging it was found that the calibration of our motion capture system was to blame. When using rigid body streaming, the number of markers and steadiness of the rigid body is paramount. When a marker doesn't show or the rigid body is unstable crazy swarm throws a 'No Updated Pose for motion capture object...'. After recalibration we are no longer getting those warnings.

whoenig, 2021-04-15:
Thanks for letting us know!

closed 2021-04-15 by whoenig

hit-JB, 2021-12-19:
@elim17 
Hello have you sole the "No updated pose" problem ,i encountered the same problem as you .Could you give me some advice how you did it? The Info is as blow
`
[ WARN] [1639884992.966357304]: No updated pose for CF cf1 for 1639884992.966298 s.
[ WARN] [1639884992.976571012]: No updated pose for CF cf1 for 1639884992.976451 s.
[ WARN] [1639884992.986326886]: No updated pose for CF cf1 for 1639884992.986274 s.
[ WARN] [1639884992.996349832]: No updated pose for CF cf1 for 1639884992.996289 s.
[ WARN] [1639884993.006453691]: No updated pose for CF cf1 for 1639884993.006391 s.
[ WARN] [1639884993.016369565]: No updated pose for CF cf1 for 1639884993.016309 s.
[ WARN] [1639884993.026424988]: No updated pose for CF cf1 for 1639884993.026348 s.
[ WARN] [1639884993.036433558]: No updated pose for CF cf1 for 1639884993.036373 s.
[ WARN] [1639884993.046464267]: No updated pose for CF cf1 for 1639884993.046405
`

whoenig, 2021-12-22:
@hit-JB: This is a closed issue and you won't get an answer here. Try the github discussions instead.