USC-ACTLab/crazyswarm #351: [crazyswarm_server-2] process has died after fresh installation

EnricaSo, 2021-03-30:
Hello!

I am having an issue since yesterday after reinstalling the `crazyswarm` project. I am following the procedure to use physical robots. After reinstalling everything successfully (I get no error), sourcing, and launching `hover_swarm.launch`, the `crazyswarm_server-2` dies with error `-11`. I am trying to figure out where the problem comes from. Could it be related to `swig`? Which is the required version?

Do you have an idea of what's going on?

EnricaSo, 2021-03-30:
The problem seems to come from the external dependency `libmotioncapture`:
```
at /home/enrica/Github/crazyswarm/ros_ws/src/externalDependencies/libmotioncapture/src/optitrack.cpp:367
367	        int nRigidBodies = 0; memcpy(&nRigidBodies, ptr, 4); ptr += 4;
```

EnricaSo, 2021-03-30:
The issue is solved, the problem was coming from the Motive configurations. I suspect that, even if unticked and inactive, having skeletons or other objects in the creation panel can cause problems.

closed 2021-03-30 by EnricaSo

whoenig, 2021-03-30:
Thanks for the update!

Sherry-97, 2021-07-14:
@EnricaSo I seem to have the same problem as you, but I can’t  to find the correct configuration of optitrack. Could you please let me know?