USC-ACTLab/crazyswarm #367: [mocap_helper-2] process has died when launching mocap_helper.launch

thibaultdesurrel, 2021-04-30:
Hello,

We are a group of students trying to use crazyswarm paired with an optitrack mocap system for a school project. We followed the configuration tutorial thoroughly but we ran into an issue when launching mocap_helper.launch.

```
thibault@thibault-OMEN-by-HP-Laptop:~/crazyswarm/ros_ws/src/crazyswarm/launch$ roslaunch mocap_helper.launch 
... logging to /home/thibault/.ros/log/141372de-a9b4-11eb-87b6-e5333552ae69/roslaunch-thibault-OMEN-by-HP-Laptop-9734.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thibault-OMEN-by-HP-Laptop:41177/

SUMMARY
========

PARAMETERS
 * /mocap_helper/motion_capture_type: optitrack
 * /mocap_helper/object_tracking_type: motionCapture
 * /mocap_helper/optitrack_host_name: optitrack
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

auto-starting new master
process[master]: started with pid [10921]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 93333866-a9b8-11eb-87b6-e5333552ae69
process[rosout-1]: started with pid [10931]
started core service [/rosout]
process[mocap_helper-2]: started with pid [10934]
frame 0:
[mocap_helper-2] process has died [pid 10934, exit code -11, cmd /home/thibault/crazyswarm/ros_ws/devel/lib/crazyswarm/mocap_helper __name:=mocap_helper __log:=/home/thibault/.ros/log/93333866-a9b8-11eb-87b6-e5333552ae69/mocap_helper-2.log].

```
Our program is configured using the hosts file in /etc, 147.250.35.30 being the ip of the pc running motive, and 127.0.1.1 our pc running ros
```
127.0.1.1	localhost
127.0.1.1	thibault-OMEN-by-HP-Laptop
147.250.35.30 optitrack
```

Our crazyflies.yaml file :
```
crazyflies:
- channel: 100
  id: 1
  initialPosition:
  - 1.5
  - 1.5
  - 0.0
  type: medium
- channel: 110
  id: 2
  initialPosition:
  - -1.0
  - -0.5
  - 0.0
  type: default
```

Does anyone know where this error might come from and how to fix it? Thanks in advance for your answers.

whoenig, 2021-04-30:
How is Motive configured? Make sure you *only* stream rigid bodies and nothing else, or have the latest version of libMotionCapture (needs to be pulled manually currently).

If that doesn't help, please report if https://github.com/whoenig/NatNetSDKCrossplatform works for you, as libMotionCapture is build around that.

thibaultdesurrel, 2021-05-04:
Hello thanks for your answer,

The motioncapture parameter was a mistake, we changed that because we want to use the ICP to have redundant marker configuration.
Our corrected mocap_helper.launch:
```
<?xml version="1.0"?>
<launch>

  <node pkg="crazyswarm" type="mocap_helper" name="mocap_helper" output="screen" >
    <rosparam>
      # tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      #vicon_host_name: "vicon"
      optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
    </rosparam>
  </node>
</launch>
```
And this is our motive settings (therefore rigid bodies at turned of)
[image]

However right now we run into this error:
```
thibault@thibault-OMEN-by-HP-Laptop:~/crazyswarm/ros_ws/src/crazyswarm/launch$ roslaunch mocap_helper.launch 
... logging to /home/thibault/.ros/log/2e3c9eea-acea-11eb-b16e-81002bd18a24/roslaunch-thibault-OMEN-by-HP-Laptop-9511.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thibault-OMEN-by-HP-Laptop:41093/

SUMMARY
========

PARAMETERS
 * /mocap_helper/motion_capture_type: optitrack
 * /mocap_helper/object_tracking_type: libobjecttracker
 * /mocap_helper/optitrack_host_name: optitrack
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

auto-starting new master
process[master]: started with pid [9519]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2e3c9eea-acea-11eb-b16e-81002bd18a24
process[rosout-1]: started with pid [9529]
started core service [/rosout]
ERROR: cannot launch node of type [crazyswarm/mocap_helper]: crazyswarm
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share
```
Do you know what might cause this? We tried to change lines 6 to 8 of libmotioncapture/CMakeLists.txt by
```
set(ENABLE_OPTITRACK True CACHE STRING "Enable Optitrack")
set(ENABLE_VICON False CACHE STRING "Enable Vicon")
set(ENABLE_PHASESPACE False CACHE STRING "Enable Phasespace")
```
but to no avail.

closed 2021-05-05 by jpreiss