USC-ACTLab/crazyswarm #369: cmdVel not implemented for simulation

Cafolkes, 2021-05-05:
Hi,

we are planning to use the cmdVel command to send attitude + thrust commands (roll, pitch, yawrate, thrust) to the crazyflies using the Python API. However, cmdVel is not implemented in the crazyflieSim class, so when running our code in simulation the crazyflie object does not have a cmdVel method. Is there a way of replicating the cmdVel functionality with methods already implemented in the crazyflieSim class or can you please give some hints on the easiest way to implement cmdVel with the simulator?

Thank you for your great work!

jpreiss, 2021-05-05:
You can look at the `cmdVelocityWorld` and `integrate` methods in `crazyflieSim.py` to get an idea of what needs to be done. It would require adding a second-order integration step using the acceleration vector derived from the (r, p, y, thrust) command.

jpreiss, 2021-07-28:
@Cafolkes did you work on this at all? If so, please consider contributing a PR. I think other users would benefit from this feature.

jpreiss, 2021-09-28:
closed due to inactivity.

closed 2021-09-28 by jpreiss