USC-ACTLab/crazyswarm #380: Better error reporting from server

jpreiss, 2021-05-10:
The Crazyswarm server sometimes fails with an uninformative error message, see e.g. https://github.com/USC-ACTLab/crazyswarm/discussions/379. More informative error messages could help reduce the Q&A load.

jpreiss, 2021-09-30:
1e1c2b1e8cdb3d1aa07c7974e6a8106fb008d779 also for this issue

jpreiss, 2021-09-30:
A big one is integer values in crazyflies.yaml `initialPosition` coordinates. This causes an `XmlRpcException`, but **not** from here:

https://github.com/USC-ACTLab/crazyswarm/blob/b4641e7ed4104054f31492b5986c44e6052a602f/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1183

I ran out of time to investigate today, but maybe the exception is happening in the rosparam node?

whoenig, 2021-09-30:
My guess is that this is from https://github.com/USC-ACTLab/crazyswarm/blob/b4641e7ed4104054f31492b5986c44e6052a602f/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1201

This should probably also allow integers and use a simple assignment in the integer case.

jpreiss, 2021-09-30:
Oh, does `ROS_ASSERT` raise an `XmlRpcException`? I would not have guessed that...

I agree that accepting integers is a better solution.

whoenig, 2021-09-30:
No, ROS_ASSERT probably doesn't raise such an explicit exception:-( Like a normal assert, it's also disabled in release builds. The problem is likely the next line, which calls a cast operator (http://docs.ros.org/en/noetic/api/xmlrpcpp/html/classXmlRpc_1_1XmlRpcValue.html#a95fa5d878b5c5946994b49f02829efc0):

https://github.com/USC-ACTLab/crazyswarm/blob/b4641e7ed4104054f31492b5986c44e6052a602f/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1202

I assume we would need a switch/if statement here and handle both the int and double case. Luckily, there are no more other (relevant) types...