USC-ACTLab/crazyswarm #381: hover_swarm.launch and mocap_helper.launch does not output Motive data

pthinesh, 2021-05-12:
I am trying to fly one Crazyflie using hover_swarm.launch with OptiTrack motion capture system. Everything works fine, except that rviz not showing the Crazyflie movements.
There were no errors in the terminal. I also tried mocap_helper.launch, there are no position updates printed.

I made the following modifications in "hover_swarm.launch" file:
      **motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker**
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      # vicon_host_name: "vicon"
      **optitrack_host_name: "192.168.1.1" # Windows (Motive) IP address**
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "192.168.1.1"
      **optitrack_local_ip: "192.168.1.3" # Ubuntu IP address
      optitrack_server_ip: "192.168.1.1" # Windows (Motive) IP address**

Motive Streaming settings:
[image]

Terminal output when running hello_world.py with hover_swarm.launch:
(Crazyflie takes off and lands without any error)
[image]

Terminal output when running mocap_helper.launch:
[image]

I'm using ROS noetic on Ubuntu 20.04.
Motive version is 2.1.2
Windows and Ubuntu are connected using a separate Ethernet port, and there are no problem with the network (motion data are successfully received when using VRPN).

whoenig, 2021-05-12:
Could you try running https://github.com/whoenig/NatNetSDKCrossplatform and report if you can receive data that way?

pthinesh, 2021-05-13:
Hi @whoenig, Thanks for the reply.

I tried 'NatNetSDKCrossplatform', but still no data received...
[image]

pthinesh, 2021-05-13:
Hi @whoenig, Thanks for the reply.

I tried 'NatNetSDKCrossplatform'; following is the output
[image]

Then again I tried VRPN (it works without any error)
[image]

pthinesh, 2021-05-13:
I tried disconnecting the Wifi and run hover_swarm.launch, following error occurred;
[image]

I can still receive data through VRPN
Could you help me resolve this?

pthinesh, 2021-05-13:
Hi @whoenig, I tried disconnecting the Wifi and run hover_swarm.launch, following error occurred;
[image]

VRPN works fine.
I am couldn't find the mistake, could you help me resolve this?

pthinesh, 2021-05-13:
Hi @whoenig, I tried disconnecting the Wifi and run hover_swarm.launch, following error occurred;
[image]

VRPN works fine.
I couldn't find the mistake, could you help me resolve this?

whoenig, 2021-05-15:
See https://github.com/whoenig/NatNetSDKCrossplatform/issues/2.

closed 2021-05-15 by jpreiss