USC-ACTLab/crazyswarm #382: When I use multiple cfs, several cfs will be unstable and start crashing.

goddogzzz, 2021-05-14:
Hello,

Recently, I used 4 drones to execute my program. When I fly each CF individually, they work fine, so I'm ruling out the fact that each drone have any defects.
However, when I start to add more CFs, they become extremely unstable and start crashing.

My crazyflies.yaml is as follows:
cf1: channel 100
cf2: channel 100
cf3: channel 110
cf4: channel 90
My hover_swarm.launch is as follows:
genericLogTopics: ["log1"]
genericLogTopicFrequencies: [1]
genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
motion_capture_type: "vrpn"
object_tracking_type: "motionCapture"
send_position_only: False
vrpn_host_name: "10.1.1.198"
save_point_clouds: ~/pointCloud.ot
print_latency: False
write_csvs: False
force_no_cache: True
enable_parameters: True
enable_logging: False

I'm using 4 reflective markers per drone. The motion capture system is tracking all 4 different rigid bodies.
The following link shows the current test run with 4 cf's.

https://youtu.be/Pp7KR1XA6Rc

Now, the entire system can achieve most of the functional effects, but it still lacks stability. What should I do to achieve more stable flight of cfs?

Thanks, hoping for some help regarding this issue.

whoenig, 2021-05-15:
This looks like it might be a tracking issue. Try setting `send_position_only` to True or try flying using `libobjecttracker` instead.

jpreiss, 2021-05-15:
I agree this is probably related to tracking. Are your marker configurations all very distinct from each other?

closed 2021-05-15 by jpreiss