USC-ACTLab/crazyswarm #396: Issue with optitrack compatability in latest crazyswarm update

williamdwarke, 2021-06-21:
Hi, I want to report that the latest update breaks optitrack compatibility. I noticed that the NatNetSDKCrossplatform tools still worked perfectly, so I compared the code with optitrack.cpp in libmotioncapture and found a bug in the for loop on line 354. It looked like the pointer offset calculation didn't match that from PacketClientHelper.cpp, so I changed 

`        
for (int i=0; i < nMarkerSets; i++)
{
    ptr += strlen(ptr) + 1;
    ptr += 12;
}
`

to 

`        
for (int i=0; i < nMarkerSets; i++)
{
    ptr += (int)strlen(ptr) + 1;
    int nMarkers = 0; memcpy(&nMarkers, ptr, 4); ptr += 4;
    for(int j=0; j < nMarkers; j++)
    {
      ptr += 12;
    }
}
`

and the problem is fixed! It may not be the most elegant solution, but hopefully this helps somebody out before the repo is updated.

jaredpaquet4ufl, 2021-06-21:
@whoenig please update the submodules this was fixed in https://github.com/USC-ACTLab/libmotioncapture/commit/c72069222404db17d23987e148b3a55bd5315784

whoenig, 2021-06-21:
Thanks to both of you!

commit 306ba4d5 (2021-06-21): Update libmotioncapture submodule

Fixes #396.
closed 2021-06-21 by whoenig