USC-ACTLab/crazyswarm #397: Crazyswarm not publishing data to all (/cf#/pose) pose messages

sabdi1, 2021-06-22:
Hello,

Currently when I run Crazyswarm, it only publishes data to the ROS messages for the last two crazyflies. For example, if I've selected crazyflie 1,3,and 4 via chooser.py, the pose messages that are being published are /cf3/pose and /cf4/pose. The topics for the pose messages for the other crazyflies are visible via rostopic list, data is just not being published to them.

Is there somewhere in the launch file or elsewhere where I may have accidentally set a restriction on the number of crazyflies or the messages to be published? 

Thanks,

whoenig, 2021-06-24:
There is no intentional restriction that I am aware of. However, the bandwidth is rather limited. Could you try changing the update from `10` to `100` in https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L662? At the same time, you may also try reducing the ping time to 1ms in https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1016.

Are all your CFs using the same radio (i.e., are they on the same channel). It might be an option to use two radios (and thus, 2 channels) instead.

sabdi1, 2021-06-30:
Hello I made the recommended alterations to the code and found that when I ran 4 cfs, 3 had their poses published and when I ran 8, 2 cfs had their poses published. 

Yes, as of right now, all cfs are using the same radio. Ideally, I'd fly all 10 cfs with the same radio. My understanding was that Crayswarm had the capability to fly up to 15 cfs per radio, hence the 45 drone display with 3 Crazyradios. Is there a separate repository with variables preset that allow for that shown capability?

whoenig, 2021-06-30:
The 15 CFs per radio are for broadcast communication only; the demonstration uses the code in this repository (although in an older version). You are using point-to-point communication, which requires a significantly higher bandwidth. We didn't do much testing for this mode and the Crazyswarm is not yet optimized for it, i.e., the scheduler is not fair, but my guess is that you can have around 4 CFs per radio, depending on the frequency etc. I am working on moving to crazyflie-link-cpp under-the-hood, which will resolve the scheduling problems (I have no concrete timeline for this, though). The new scheduler should avoid issues you are seeing, that some CFs do not receive any data.

uhvardhan, 2022-01-10:
Hello, 

I was wondering if you were able to find a solution for this. I want to control 8 CFs using 3 different radios. Let's say:
CFs 1,2,3 -> Channel 120
CFs 4,5,6 -> Channel 110
CFs 7,8 -> Channel 100
I am currently using Loco Positioning System, so I would like to save the poses of all CFs in a bag file. This should theoretically work right - as in I should be able to do it?

sabdi1, 2022-02-02:
I don't see why not. As long as you've enabled logging in the launch file, have enough radios, and have the allCrazyflies.yaml set properly, I don't see a reason this wouldn't be possible.