USC-ACTLab/crazyswarm #448: cfclient and crazsywarm numpy version issue

The-SS, 2021-08-04:
There might be a numpy version mismatch issue between cfclient and crazswarm. 
cfclient requires numpy 1.19 (it fails to start with newer versions) while crazyswarm simulations fail with numpy 1.19 (they worked when I upgraded to 1.21).

whoenig, 2021-08-04:
what is the failure case here?

The-SS, 2021-08-04:
cfclient fail error with numpy 1.21: 
```pkg_resources.ContextualVersionConflict: (numpy 1.21.1, Requirement.parse('numpy~=1.19.2'), {'cfclient'})```

crazyswarm simulator error with numpy 1.19: 
```ImportError: numpy.core.multiarray failed to import``` 
when running `python figure8_csv.py --sim`

jpreiss, 2021-08-04:
hmm, 1.19 is only one year old. Since Crazyswarm tests also run on Python 2.7 I don't think they should require such a new version of NumPy.

It's still weird that cfclient is pinned to 1.19 but I would expect our tests to work with 1.19...

The-SS, 2021-08-04:
I am on Ubuntu 20.04 (so I'm using python3). I did not select numpy 1.21.1, I just used `pip install -U numpy`.

jpreiss, 2021-08-04:
Yes, for that reason I think cfclient loosening their version spec would be the best way to fix this issue.

But if for some reason they are stuck with 1.19, it should not cause Crazyswarm to fail.

In general, we want Crazyswarm to work with many different versions of NumPy. Typical Crazyswarm scripts use a lot of other numerical packages, and we want to make sure the dependency resolution problem always has a solution.

whoenig, 2021-09-27:
The restrictive versioning was fixed on the cfclient side. Is there anything we need to do on our side? I have numpy 1.19.5 and don't see any problem on my side, so I am voting for closing this issue.

closed 2021-09-27 by The-SS