USC-ACTLab/crazyswarm #461: Race in compilation?

jonasdn, 2021-08-17:
Hi!

In the repository [crazyflie-testing](https://github.com/bitcraze/crazyflie-testing) we build and test Crazyswarm each night, occasionally it will fail.

It did today, and ten days ago:

- [today](https://github.com/bitcraze/crazyflie-testing/actions/runs/1137929973)
- [10 days ago](https://github.com/bitcraze/crazyflie-testing/actions/runs/1107191140)

When it fails it fails in compilation with:

```
[ 94%] Generating C++ code from crazyswarm/NotifySetpointsStop.srv
[ 94%] Generating C++ code from crazyswarm/SetGroupMask.srv
/__w/crazyflie-testing/crazyflie-testing/crazyswarm/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp:8:10: fatal error: crazyswarm/UpdateParams.h: No such file or directory
    8 | #include "crazyswarm/UpdateParams.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Although nothing seem to have changed in the source between runs. Is there some kind of race between generation and compilation perhaps?

whoenig, 2021-08-17:
There is certainly a dependency issue that we `fixed` here: https://github.com/USC-ACTLab/crazyswarm/blob/master/build.sh#L27-L29. Since you are just calling `build.sh`, I guess we would actually need to try to fix the actual dependency in the catkin build system.

jonasdn, 2021-08-17:
Ok! What is the option to "just calling `build.sh`" ?

whoenig, 2021-08-17:
That's the right approach - sorry for the confusion. I'll keep this open as a bug here and try to fix the dependency properly!

whoenig, 2021-08-17:
I hope that this is fixed in master now. @jonasdn, please reopen this issue if you see this again as we didn't see it in our CI and I was unable to reproduce the problem on a parallel build with 24 cores.

closed 2021-08-17 by whoenig

jonasdn, 2021-08-17:
Thanks Wolfgang!