USC-ACTLab/crazyswarm #463: Python APIs are not shown in the document

loujiabin1994, 2021-08-24:
Hello, these days I want to look up the python API of crazyflie class, but it's not shown in docs. 
       The address is https://crazyswarm.readthedocs.io/en/latest/api.html#crazyflie-class.

The-SS, 2021-08-24:
I noticed this as well. 
While it may be less convenient, the comments/docstrings in [crazyflie.py](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py) are basically what you are looking for.

jpreiss, 2021-08-24:
Should be an easy fix.

It's annoying that our docs build in CI shows the message

    WARNING: autodoc: failed to import class 'crazyflie.CrazyflieServer' from module 'pycrazyswarm';
    the following exception was raised:
    No module named 'numpy'

but doesn't return a nonzero exit code. Maybe we can configure autodoc differently or parse the output.

jpreiss, 2021-09-28:
Fixed by #480 and 73cce6b. Changed warnings to errors in docs CI.

closed 2021-09-28 by jpreiss