USC-ACTLab/crazyswarm #505: vrpn/what(): Unknown motion capture type !

wangmenghetrue, 2021-10-17:
Hello!The motion capture system I use is another type, but it requires vrpn.WhenI use the vrpn of the motion capture system to communicate with crazyswarm.`roslaunch crazyswarm hover_swarm.launch`It will appear `what( ): Unknown motion capture type !`
Ubuntu version is ubuntu 20.04.
    My local `hover_swarm.launch` file is follow:
`# tracking`
`motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn`
`object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker`
`send_position_only: False # set to False to send position+orientation; set to True to send position only`
`#vicon_host_name: "vicon"`
` # optitrack_host_name: "optitrack"`
`# qualisys_host_name: "10.0.5.219"`
`# qualisys_base_port: 22222`
`vrpn_host_name: "192.168.2.137"`
`save_point_clouds: ~/pointCloud.ot`
`print_latency: False`
`write_csvs: False`
` force_no_cache: False`
I also try to run `mocap_helper.launch`,but its result is still `what( ): Unknown motion capture type !`
I don't known what i should do.I really hope to get your help! Thank you very much!

whoenig, 2021-10-18:
Are you using the latest version of the Crazyswarm (specify the commit and make sure you updated all submodules, including `git submodule sync`)? We had VRPN disabled for a while, but it is enabled again by default, see https://github.com/IMRCLab/libmotioncapture/blob/651bbc22c95e282d962ed39e52f0988b65ac7892/CMakeLists.txt#L9.

wangmenghetrue, 2021-10-18:
Thank you so much for replying to me! I found that my crazyswarm is not the latest version. I reinstalled the latest version of crazyswarm as you suggested. But its `hover_swarm.launch` hasn't`vrpn_host_name:""`.I don't know how to set IP.
[image]
When I run `hover_swarm.launch`, it shows
`[ INFO] [1634562839.970200116]: libmotioncapture connecting to vrpn at hostname ' ' - might block indefinitely if unreachable!`
`vrpn_connect_udp_port: error finding host by name ().`
`vrpn_connection_IP: Can't set up socket to lob UDP packets!`
[image]
I think the reason may be that `vrpn_host_name: " "` is not set in `hover_swarm.launch`.But its `hover_swarm.launch` hasn't`vrpn_host_name:""`.What should i do?