USC-ACTLab/crazyswarm #519: Cannot find -lNatNet

The-SS, 2021-11-01:
Per [https://github.com/whoenig/NatNetSDKCrossplatform/issues/3](https://github.com/whoenig/NatNetSDKCrossplatform/issues/3), I cloned crazyswarm and pulled `crazyswarm/ros_ws/src/crazyswarm/externalDependencies/libmotioncapture` recursively. I changed [ENABLE_OPTITRACK_CLOSED_SOURCE](https://github.com/IMRCLab/libmotioncapture/blob/13d14f85d29171426a2b663cf71cb6262c6f7760/CMakeLists.txt#L8) to ON and was able to build `libmotioncapture`. 
However, when building with crazyswarm's `./build.sh` script, the package is failing to find NatNet.
```
[ 98%] Linking CXX executable /home/mobilerobot/Crazyflie_robots/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server
[ 98%] Linking CXX executable /home/mobilerobot/Crazyflie_robots/crazyswarm/ros_ws/devel/lib/crazyswarm/mocap_helper
/usr/bin/ld: cannot find -lNatNet
collect2: error: ld returned 1 exit status
make[2]: *** [crazyswarm/CMakeFiles/mocap_helper.dir/build.make:209: /home/mobilerobot/Crazyflie_robots/crazyswarm/ros_ws/devel/lib/crazyswarm/mocap_helper] Error 1
make[1]: *** [CMakeFiles/Makefile2:1278: crazyswarm/CMakeFiles/mocap_helper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lNatNet
collect2: error: ld returned 1 exit status
make[2]: *** [crazyswarm/CMakeFiles/crazyswarm_server.dir/build.make:209: /home/mobilerobot/Crazyflie_robots/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:1312: crazyswarm/CMakeFiles/crazyswarm_server.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
I'm not sure what else is missing here to get crazyswarm to compile. Any other modifications that come to mind?

whoenig, 2021-11-01:
Try adding

```
link_directories(
${LIBMOTIONCAPTURE_LINK_DIR}
)
```

at https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/CMakeLists.txt#L155.

I can look into a proper integration (building by default) towards the end of this week. It's not enabled by default, because this would break the Windows and Mac build of libmotioncapture, but since the crazyswarm only supports Ubuntu, we could enable it there always.

The-SS, 2021-11-01:
That worked. Thanks!

whoenig, 2021-11-01:
Great. I also made a proper fix, see https://github.com/USC-ACTLab/crazyswarm/pull/520. This will merge within a couple of days, hopefully.

The-SS, 2021-11-02:
Regarding this, I wanted to clarify that the package _compiled_ with the closed-source option.
 _However_, I was still having issues with mocap_helper getting the optitrack data. I noticed that mocap_helper was only getting point cloud data when a rigid body was created for the crazyflie; i.e. I would have to create rigid bodies to use Crazyswarm's ICP-based tracking (which makes no sense). Labeled and unlabeled markers were both on in Motive 2.3. When  I would check the cf rigid body, mocap_helper would receive the point cloud data; when I would uncheck it, mocap_helper would stop receiving the point cloud data. 

I switched from `optitrack_closed_source` to the open-source `optitrack` implementation and now it worked properly. I don't know exactly what changed from my last comment in [NatNetSDKCrossplatform issue 3](https://github.com/whoenig/NatNetSDKCrossplatform/issues/3), but the open-source solution seems to be working now.

whoenig, 2021-11-03:
Thanks for the feedback! I didn't have access to try it yet with the Crazyswarm, so I only focused on the linking issues. I am re-opening this issue to test (&fix) this on my side.
Glad to hear that you are unblocked for now, though!

The-SS, 2021-11-03:
So the machine I'm using has two ethernet ports, one for OptiTrack and another for internet.
When the open-source solution was used, crazyswarm was defaulting to the ethernet port hooked up to the internet instead of that connected to OptiTrack. 
When I tested with NatNetSDKCrossplatform, the packetClient (open source) did that as well while the sampleClient (closed source) was defaulting to the ethernet port connected to OptiTrack. Thus, the open source would not get data while the closed source would get data.

For now, I'll just have to turn off the internet connection and use the open-source solution.

whoenig, 2021-11-03:
Interesting finding. I think you would need to change https://github.com/IMRCLab/libmotioncapture/blob/main/src/optitrack.cpp#L317 to use the IP address of the Ethernet adapter that you want to listen to (here: the local IP associated with the Ethernet adapter that connects to your optitrack machine). From what I remember, this is how boost decides which network interface to use. If that's the case, I could add an option to change this value without recompiling.

The-SS, 2021-11-03:
When using NatNetSDKCrossplatform I tried setting [https://github.com/whoenig/NatNetSDKCrossplatform/blob/b1a68bb13fa3bab92ce584e5e295b80229399922/src/main.cpp#L111](https://github.com/whoenig/NatNetSDKCrossplatform/blob/b1a68bb13fa3bab92ce584e5e295b80229399922/src/main.cpp#L111) to the desired IP but that didn't work. 
I also tried changing the line you noted in libmotioncapture but that didn't work either for me.

closed 2021-11-04 by jpreiss