USC-ACTLab/crazyswarm #529: Cannot control drone using OptiTrack mocap system

sajisaka, 2021-11-13:
Hello, everyone,

I try to control Crazyflie by a motion capture (OptiTrack + Motive + VRPN + ROS melodic).
I make a frame to attach markers, and I tried a flight according to instruction of crazyswarm, but a drone does not seem to be controlled.

This is my result.
https://youtu.be/735klXdhw2Y
(Lain on my back immediately without taking off)

I confirm the following dots
1) I set the Z-axis on the top in Motive
2) I confirm that a drone can take off by manual manipulation with marker frame
3) I confirm that the position of the drone is recognized on Rviz

Is there anything to be checked?

Regards,
AJISAKA
Top

sajisaka, 2021-11-13:
I changed MoCap frame to official deck and try again, but there is still problem
https://youtu.be/x_i-MFmzQd0

whoenig, 2021-11-13:
The video looks like a coordinate system mismatch. You should make sure that if you move the drone manually with your hand, that positive x is front (where the antenna is) and positive why is towards the left.

Are you indeed using VRPN rather than NatNet? Have you tried using optitrack or optitrack_closed_source instead?

sajisaka, 2021-11-14:
Thank you whoenig

I checked coordinate and it locks like not bad
https://youtu.be/inerQZ6QIek
this is coordinate check using lighthouse position system and it works perfectly.
(I can control my crazyflie automatically)

https://youtu.be/ciLvXS4sdKs
And this is same check using Mocap.
It looks like same as lighthouse, but this time I couldn't control.

>Are you indeed using VRPN rather than NatNet? Have you tried using optitrack or optitrack_closed_source instead?
I tried to use Optitrack but it works something wrong (crazyswarm receives nothing, is there any tutorial?)

Regards

sajisaka, 2021-11-14:
I found that "/cf1/pose" topic is incorrect even if "cf1" frame position is OK

Here is the result.
https://youtu.be/VGUnXRUv-F4

"/cf1/pose" is not same as "cf1" frame position
(It looks like increase position and suddenly reset to 0)

How can I fix it?

whoenig, 2021-11-14:
Good catch! This typically means that the broadcasts are not working properly. Did you update your Crazyradio firmware as described in the documentation? This is required even if you recently bought the radio.

NatNet vs VRPN: There is no tutorial besides the Crazyswarm documentation. You can search the issues and discussion for a few hints, though. Basically, you want to first try `mocap_helper.launch` or NatNetSDKCrossplatform for troubleshooting. Feel free to open a new discussion for any issues with that. NatNet is the preferred & tested mode. VRPN does not have all features and is rarely tested on our side.

sajisaka, 2021-11-15:
Dear  whoenig,

I updated my crazyradio to ver 99.55 but the situation become more worse :

cfclient doesn't recognize my crazyradio, and I cannot connect to my crazyflie via crazyradio.

"rosrun crazyflie_tools scan -v" reports there is a crazyradio ver 99.55
but "rosrun crazyflie_tools scan --address E7E7E7E701" reports nothing.
(of course, I replug crazyradio after update firmware, and I set ID to E7E7E7E701 via USB)

Regards

closed 2021-11-16 by jpreiss