USC-ACTLab/crazyswarm #538: Build error in crazyflie-testing action

jonasdn, 2021-11-22:
It _seems_ that the commit: dce6c0c460957f6bf1f3c0915d7c35a3952f163f

Caused a build error in our action: https://github.com/bitcraze/crazyflie-testing/runs/4129139144?check_suite_focus=true

```
[ 75%] Linking CXX shared library /__w/crazyflie-testing/crazyflie-testing/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so
/usr/bin/ld: skipping incompatible /__w/crazyflie-testing/crazyflie-testing/crazyswarm/ros_ws/src/crazyswarm/externalDependencies/libmotioncapture/deps/NatNetSDKCrossplatform/lib/ubuntu/libNatNet.so when searching for -lNatNet
/usr/bin/ld: cannot find -lNatNet
collect2: error: ld returned 1 exit status
make[2]: *** [crazyswarm/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:181: /__w/crazyflie-testing/crazyflie-testing/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3488: crazyswarm/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

jonasdn, 2021-11-22:
Might be because this build is on ARM? (Raspberry PI)

jonasdn, 2021-11-22:
Is there a way to opt out of this stuff for the crazyflie-testing build?

whoenig, 2021-11-22:
This should be fixed in 4d6ca47 (for some users we link against a closed source library, where we only have binaries for x64 architectures; this is hopefully now disabled on ARM). Please re-open if the push didn't fix your problem!
Thanks for reporting.

commit 4d6ca47b (2021-11-22): Enable optitrack_closed_source only on 64-bit Platforms

Fixes #538.
closed 2021-11-22 by whoenig