USC-ACTLab/crazyswarm #539: test_ros.py failure on PyKDL import

PartiallyObservable, 2021-11-23:
Hello, 

 I am attempting to install crazyswarm (for physical robots) on Ubuntu 20.04 with python 3.7.11 but I am encountering a "test_ros" failed test during PyKDL import. The full message is here:

`
=================================== FAILURES ===================================
_______________________________ test_ros_import ________________________________

    @pytest.mark.ros
    def test_ros_import():
>       from pycrazyswarm.crazyflie import TimeHelper, CrazyflieServer

test_ros.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
pycrazyswarm/crazyflie.py:9: in <module>
    import tf_conversions
/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/__init__.py:30: in <module>
    from .posemath import *
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

    from geometry_msgs.msg import Pose, Point, Quaternion
    from tf import transformations
    import tf
    import rospy
    import numpy
    
>   from PyKDL import *
E   ModuleNotFoundError: No module named 'PyKDL'

/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/posemath.py:34: ModuleNotFoundError

`

The error says that PyKDL has not been installed, however attempting to install PyKDL with the obvious 'pip install PyKDL' results in this error:

`Collecting PyKDL
  Using cached PyKDL-1.4.0.tar.gz (1.5 kB)
  Preparing metadata (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/mark/.pyenv/versions/3.7.12/envs/crazyswarm/bin/python3.7 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n5sa5xgo/pykdl_3c3834b118ba4e4fb7978f9c331a0ded/setup.py'"'"'; __file__='"'"'/tmp/pip-install-n5sa5xgo/pykdl_3c3834b118ba4e4fb7978f9c331a0ded/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-zrlv5z1w
       cwd: /tmp/pip-install-n5sa5xgo/pykdl_3c3834b118ba4e4fb7978f9c331a0ded/
  Complete output (7 lines):
  running egg_info
  creating /tmp/pip-pip-egg-info-zrlv5z1w/PyKDL.egg-info
  writing /tmp/pip-pip-egg-info-zrlv5z1w/PyKDL.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-zrlv5z1w/PyKDL.egg-info/dependency_links.txt
  writing top-level names to /tmp/pip-pip-egg-info-zrlv5z1w/PyKDL.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-zrlv5z1w/PyKDL.egg-info/SOURCES.txt'
  error: package directory 'PyKDL' does not exist
  ----------------------------------------
`

As an attempt to fix this I looked up PyKDL package on [PyPi](https://pypi.org/project/PyKDL/1.4.0/#files) and it appears that PyKDL only has a python 3.6 package, so I followed the install instructions again, but this time with python 3.6. This leads to an undefined symbol error

`
_______________________________ test_ros_import ________________________________

    @pytest.mark.ros
    def test_ros_import():
>       from pycrazyswarm.crazyflie import TimeHelper, CrazyflieServer

test_ros.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
pycrazyswarm/crazyflie.py:9: in <module>
    import tf_conversions
/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/__init__.py:30: in <module>
    from .posemath import *
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

    from geometry_msgs.msg import Pose, Point, Quaternion
    from tf import transformations
    import tf
    import rospy
    import numpy
    
>   from PyKDL import *
E   ImportError: /home/mark/.pyenv/versions/3.6.11/envs/cs/lib/python3.6/site-packages/PyKDL.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN3KDL4TreeC1ERKSs

/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/posemath.py:34: ImportError

`

I'm not 100% sure this test is necessary to pass, or if I should instead ask this question on a ROS forum, but thank you for any help!

(as a side note, I also fail "test_videoOutput" test, but I believe that is an optional feature that I don't really need, the failure is that ffmpeg doesn't have attirubte 'probe' on line 67).

whoenig, 2021-11-24:
This seems more like a ROS problem rather than a Crazyswarm issue. Have you tried installing ros-noetic-tf2-kdl or python3-pykdl?

PartiallyObservable, 2021-12-03:
I think you are correct that this is an issue with ROS, or with how ROS interacts with python. 

I was attempting to use a virtual environment (managed by pyenv) to install crazyswarm python dependencies. I've used ROS in the past with virtualenvs without any issue so I am unsure why this particular PyKDL package is causing problems.

I've simply resorted to using the system python (3.8) for now and I no longer have an issue with PyKDL and I pass all of the tests (except the video output one). Since this issue has nothing to do with crazyswarm I think it can be closed. Thank you for your help though :)

closed 2021-12-03 by whoenig