USC-ACTLab/crazyswarm #582: candidate release 2022.01 crazyflie-firmware logging issue

knmcguire, 2022-01-25:
Hi all!

We are currently preparing a candidate release for the [ crazyflie-firmware](https://github.com/bitcraze/crazyflie-firmware) and I noticed when I do hover_swarm.launch (with Lighthouse, `motion_capture_type to none `and    `enable_logging: True` &  `enable_logging_pose: True` ), I get the following error from logging:

```
[ INFO] [1643120444.533064839]: [cf1] Requesting parameters...
[ INFO] [1643120444.561687072]: Found variables in cache.
[ INFO] [1643120444.852392058]: [cf1] reqParamTOC: 0.374772 s (232 params)
[ INFO] [1643120444.852420946]: [cf1] Requesting logging variables...
[ INFO] [1643120444.855290664]: Found variables in cache.
[ INFO] [1643120444.855770919]: [cf1] reqLogTOC: 0.003381 s
[ INFO] [1643120444.860398258]: [cf1] logBlocks: 0.374772 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find stateEstimate.y in log toc!
[crazyswarm_server-2] process has died [pid 76192, exit code -6, cmd /home/kimberly/Development/ROS/ros/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/kimberly/.ros/log/fac80be0-7de9-11ec-8a46-61f8d39feaf5/crazyswarm_server-2.log].
log file: /home/kimberly/.ros/log/fac80be0-7de9-11ec-8a46-61f8d39feaf5/crazyswarm_server-2*.log
```

Once I downgraded the crazyflie to 2021.06, this error does not occur. But stateEstimate.y does exist in both versions. I'm not sure what this could have caused it. 

I just want to notify @jpreiss @whoenig since you wanted the latest release to work with Crazyswarm.

whoenig, 2022-01-25:
Thanks for the heads-up! We had a few reports on this with other firmware versions too and I assume it is a bug somewhere in crazyflie_cpp. I've been using the latest firmware with Crazyswarm2 (which uses the new cflink native backend) without issues, so I doubt that a fix will require a firmware change.

knmcguire, 2022-01-26:
ah oke, good to know. I've havent tried Crazyswarm2 yet with the new release but I will need to work from home the rest of the week. Anyway, would be good to just double test very latest of the firmware on your setup if you have any time this week :)