USC-ACTLab/crazyswarm #606: Hi, I am currently starting to work with crazyswarm and crazyflies(2.1).

GideonLombardo, 2022-03-03:
Hi, I am currently starting to work with crazyswarm and crazyflies(2.1).

I installed and configured everything, but I have big troubles with the first hovering with HW.

Everytime I execute the hello_world.py, the crazyflie flips over or flies somewhere. 
I already tried to balance everything and also tried to start the drone manually (which is working completely fine)

I also tried to change the drone, but the issue remains.

Is it possible, that I made a mistake, when setting up crazyswarm?

Thank you in advance!

Regards Florian

_Originally posted by @flrnhbr1 in https://github.com/USC-ACTLab/crazyswarm/discussions/603_

I have this same issue. My error message is as follows:

Warning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.

cfg = yaml.load(crazyflies_yaml)

Warning: No joystick found!

Traceback (most recent call last):

File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 567, in connect

self.socket.connect((dest_addr, dest_port))

ConnectionRefusedError: [Errno 111] Connection refused

I believe it might be an issue with the radio communication. I cannot determine whether the issue is on the radio's end or the crazyflie's end.

whoenig, 2022-03-08:
Please see the discussion in https://github.com/USC-ACTLab/crazyswarm/discussions/603. If your problem is unrelated, open a new discussion topic.

closed 2022-03-08 by whoenig

GideonLombardo, 2022-03-09:
I will take a look. Thank you for all your help and patience as I try to get this up and running for my lab. I am learning as I go, so it really helps to have a support structure like the discussions on Github.